orbbec/pyorbbecsdk #12: Windows python SDK 安装完之后运行案例报错

zxecho, 2023-11-12:
如题，按照教程全程无问题编译安装完sdk，运行color_viewer.py案例报错：
[11/12 15:17:14.314609][info][38524][Context.cpp:67] Context created with config: xxxxxx\orbbec\Gemini2\pyorbbecsdk\config\OrbbecSDKConfig_v1.0.xml
[11/12 15:17:14.314719][info][38524][Context.cpp:72] Context work_dir=xxxxxx\orbbec\Gemini2\pyorbbecsdk\examples
[11/12 15:17:14.386905][info][38524][DeviceManager.cpp:15] Current found device(s): (1)
[11/12 15:17:14.387088][info][38524][DeviceManager.cpp:24]      - Name: DaBai DCL, PID: 0x0701, SN/ID: AUGS83P005P, Connection: USB2.0
[11/12 15:17:14.387389][info][38524][Pipeline.cpp:15] Try to create pipeline with default device.
[11/12 15:17:14.484060][error][38524][WinHelpers.cpp:83] MFCreateDeviceSource(deviceAttrs_, &deviceSource_) returned: HResult 0x80070005: "拒绝访问。"
[11/12 15:17:14.484185][warning][38524][ObException.hpp:40] MFCreateDeviceSource(deviceAttrs_, &deviceSource_) returned: HResult 0x80070005: "拒绝访问。"
[11/12 15:17:14.484317][warning][38524][Gemini2Device.cpp:296] Execute failure! A libobsensor_exception has occurred!
         - where:296#libobsensor::Gemini2Device::createCommand
         - msg:MFCreateDeviceSource(deviceAttrs_, &deviceSource_) returned: HResult 0x80070005: "拒绝访问。"
         - type:class libobsensor::windows_pal_exception
[11/12 15:17:19.500999][warning][38524][MessengerLibusb.cpp:31] control_transfer returned error, index:0, error:No such file or directory, number:2
[11/12 15:17:19.501273][error][38524][HostProtocol.cpp:947] get struct data failed. propertyId: 1043, rc: {statusCode: 2, respErrorCode: 65535, msg: receive control transfer failed!}
[11/12 15:17:19.501396][warning][38524][ObException.hpp:40] Request failed, statusCode: 2, msg: receive control transfer failed!
Traceback (most recent call last):
  File "xxxxxxx/orbbec/Gemini2/pyorbbecsdk/examples/color_viewer.py", line 66, in <module>
    main()
  File "xxxxxxx/orbbec/Gemini2/pyorbbecsdk/examples/color_viewer.py", line 30, in main
    pipeline = Pipeline()
RuntimeError: Caught an unknown exception!
[11/12 15:17:19.721303][info][38524][Context.cpp:83] Context destroyed

还请指点一下如何解决

hzcyf, 2023-11-13:
系统MediaFoundation库返回「拒绝访问」 错误， 可能：
1. 是不是设备已经被其他应用打开了？
2. 是不是有安装一些安全软件限制了USB设备/摄像头访问权限？

另外，可以在相同条件下尝试使用OrbbecViewer看下能不能打开设备。

luzhensss, 2025-11-15:
python SDK 安装完之后运行案例报错(yolov8) @-virtual-machine:~/Desktop/orbbec/pyorbbecsdk-2-main/examples$ python color.py 
load extensions from /home/lu/miniconda3/envs/yolov8/lib/python3.10/site-packages/extensions
[11/15 21:12:08.361479][error][28395][UsbEnumeratorLibusb.cpp:157] Invalid descriptor index: 0
[11/15 21:12:08.361512][error][28395][UsbEnumeratorLibusb.cpp:403] Failed to query USB device serial number
[11/15 21:12:08.365583][error][28395][UsbEnumeratorLibusb.cpp:157] Invalid descriptor index: 0
[11/15 21:12:08.365599][error][28395][UsbEnumeratorLibusb.cpp:403] Failed to query USB device serial number
[11/15 21:12:08.372385][error][28395][UsbEnumeratorLibusb.cpp:157] Invalid descriptor index: 0
[11/15 21:12:08.372409][error][28395][UsbEnumeratorLibusb.cpp:419] Failed to query USB device interface name
[11/15 21:12:09.379434][error][28395]