orbbec/pyorbbecsdk #18: 视频流保存为视频时丢帧严重，达不到设置的30帧。

tensor-song, 2023-12-18:
视频流保存为视频时丢帧严重，达不到设置的30帧。

jian-dong, 2023-12-21:
@tensor-song 您好，您用的是什么相机设置的那个分辨率呢。 cc @zhonghong322

tensor-song, 2023-12-21:
Gemini 2L   1280*720 fps=30
| |
宋
|
|
***@***.***
|
---- 回复的原邮件 ----
| 发件人 | Joe ***@***.***> |
| 发送日期 | 2023年12月21日 10:28 |
| 收件人 | ***@***.***> |
| 抄送人 | ***@***.***> ,
***@***.***> |
| 主题 | Re: [orbbec/pyorbbecsdk] 视频流保存为视频时丢帧严重，达不到设置的30帧。 (Issue #18) |

@tensor-song 您好，您用的是什么相机设置的那个分辨率呢。 cc @zhonghong322

zhonghong322, 2023-12-21:
您好，你的PC配置是怎样、开启了 哪些流，都是什么格式、帧率。 比如Depth、IR、Color。 格式： mjpeg，Y8，Y16，帧率： 30fps，15fps。

tensor-song, 2023-12-21:
code 基本精简没啥东西了，只开了color流，格式是默认的，帧率30fps
PC：Intel(R) Core(TM) i5-7500 CPU @ 3.40GHz   3.40 GHz   内存：32.0 GB (31.9 GB 可用)
目前保存的视频至少丢一半，麻烦看看是不是代码有问题
代码如下：
FPS = 30
FRAME_SIZE = 1280

SAVE_ROOT = 'G:\slc\camera\output'
fourcc = cv2.VideoWriter_fourcc(*'XVID') #('M', 'J', 'P', 'G') #

def main():
    pipeline = Pipeline()
    config = Config()

    profile_list = pipeline.get_stream_profile_list(OBSensorType.COLOR_SENSOR)
    color_profile: VideoStreamProfile = profile_list.get_video_stream_profile(FRAME_SIZE, 0, OBFormat.RGB, FPS)
    config.enable_stream(color_profile)

    out_color = cv2.VideoWriter(os.path.join(SAVE_ROOT, f'color_{0}.avi'), fourcc, color_profile.get_fps(),
                                (color_profile.get_width(), color_profile.get_height()))
    pipeline.start(config)
    n = 1
    while True:
        try:
            frames: FrameSet = pipeline.wait_for_frames(int(100))
            if frames is None:
                print('frame is none')
                continue

            color_frame = frames.get_color_frame()
            if color_frame is None:
                print('color frame is None')
                continue

            if color_frame is not None:
                color_image = frame_to_bgr_image(color_frame)
                out_color.write(color_image)
                print(n)
                n += 1

        except KeyboardInterrupt:
            break
    pipeline.stop()

if __name__ == "__main__":
    main()

jian-dong, 2023-12-21:
@tensor-song 你好
我稍微改了一下您的代码，测试了一下帧率应该是够的，你看一下
```python
def main():
    pipeline = Pipeline()
    config = Config()

    profile_list = pipeline.get_stream_profile_list(OBSensorType.COLOR_SENSOR)
    color_profile: VideoStreamProfile = profile_list.get_video_stream_profile(FRAME_SIZE, 0, OBFormat.RGB, FPS)
    config.enable_stream(color_profile)

    out_color = cv2.VideoWriter(os.path.join(SAVE_ROOT, f'color_{0}.avi'), fourcc, color_profile.get_fps(),
                                (color_profile.get_width(), color_profile.get_height()))
    pipeline.start(config)
    n = 0
    last_time = 0
    while True:
        try:
            frames: FrameSet = pipeline.wait_for_frames(int(100))
            if frames is None:
                print('frame is none')
                continue

            color_frame = frames.get_color_frame()
            if color_frame is None:
                print('color frame is None')
                continue

            if color_frame is not None:
                color_image = frame_to_bgr_image(color_frame)
                out_color.write(color_image)
                n += 1
                if last_time == 0:
                    last_time = time.time()
                else:
                    if time.time() - last_time > 1:
                        print(f'FPS: {n / (time.time() - last_time)}')
                        n = 0
                        last_time = time.time()

        except KeyboardInterrupt:
            break
    pipeline.stop()

if __name__ == "__main__":
    main()
```