orbbec/pyorbbecsdk #20: raise InitializationError("OpenNI2 could not be loaded:\n    %s" %   openni.utils.InitializationError: OpenNI2 could not be loaded:     D:\code_2\code\OpenNI2.dll: file does not exist`

sun-rabbit, 2024-01-02:
你好，我按照https://zhuanlan.zhihu.com/p/563844587?utm_id=0，进行驱动astro，gemini，但是显示
`  File "d:\code_2\code\fruit\test_depth.py", line 45, in <module>       
    openni2.initialize()#初始化openni库
  File "C:\Users\PT\AppData\Local\Programs\Python\Python310\lib\site-packages\openni\openni2.py", line 104, in initialize
    raise InitializationError("OpenNI2 could not be loaded:\n    %s" %  
openni.utils.InitializationError: OpenNI2 could not be loaded:
    D:\code_2\code\OpenNI2.dll: file does not exist`
错误，我询问了技术支持，但没有给我相应的解答是为什么？三天前这个摄像头还是可以正常测试的，但是今天测试时候就出现了这个问题，我没有更改代码和任何环境变量

jian-dong, 2024-01-02:
@sun-rabbit 您好，这个库和那个知乎链接的库没有关系的，请按照这这个repo README里面的步骤，不要去参考那个知乎链接。

sun-rabbit, 2024-01-03:
> @sun-rabbit 您好，这个库和那个知乎链接的库没有关系的，请按照这这个repo README里面的步骤，不要去参考那个知乎链接。
我没有使用这个库，我当时编译太慢了，请问我在进行深度检测时候，为什么有时候我从深度流里拿出来的深度图片全是0

sun-rabbit, 2024-01-03:
> @sun-rabbit您好，这个库和那个知乎链接的库没有关系的，请按照这个库 README 里面的步骤，不要去参考知乎链接。

如果我在上位机（windows上）常用python，我编译python版本还是windows版本呢

jian-dong, 2024-01-03:
> > @sun-rabbit 您好，这个库和那个知乎链接的库没有关系的，请按照这这个repo README里面的步骤，不要去参考那个知乎链接。
> > 我没有使用这个库，我当时编译太慢了，请问我在进行深度检测时候，为什么有时候我从深度流里拿出来的深度图片全是0

有日志么，看看

jian-dong, 2024-01-03:
> > @sun-rabbit您好，这个库和那个知乎链接的库没有关系的，请按照这个库 README 里面的步骤，不要去参考知乎链接。
> 
> 如果我在上位机（windows上）常用python，我编译python版本还是windows版本呢

参考windows版本的readme