orbbec/pyorbbecsdk #21: 如何同步多摄像头

TitansWhale, 2024-01-15:
我在使用多个摄像头同步录制数据（未使用同步线），我希望同步多个摄像头的时间戳，在viewer里面我看到有个同步摄像头时间戳。在python中应该如何实现这个功能。
[image]
我在pyi里面搜索“syn”里面找到了    

```
def enable_multi_device_sync(self, arg0: int) -> None:
        """
        Activates the multi-device synchronization function to synchronize the clock of the created device (the device needs to support this function).repeat_interval: The synchronization time interval (unit: ms; if repeatInterval=0, it means that it will only be synchronized once and will not be executed regularly).
        """

```
这个函数能否实现多摄像头同步功能？

jian-dong, 2024-01-17:
参考一下[这个实例](https://github.com/orbbec/pyorbbecsdk/blob/main/examples/two_devices_sync.py)

tushardmaske, 2024-01-30:
File "/home/tmaske/Tushar/office/camera_setup/orbbec/pyorbbecsdk_new/pyorbbecsdk/examples/two_devices_sync.py", line 178, in main
    sync_config_json = multi_device_sync_config[serial_number]
KeyError: 'CL8S93P007M'

hplegend, 2024-05-18:
> 参考一下[这个实例](https://github.com/orbbec/pyorbbecsdk/blob/main/examples/two_devices_sync.py)

我参考这个例子跑，但是似乎无法同步，第二个设备无法返回数据，导致只能渲染第一个设备的彩色图像和深度图像

hplegend, 2024-05-18:
> 参考一下[这个实例](https://github.com/orbbec/pyorbbecsdk/blob/main/examples/two_devices_sync.py)

再多问一句哈，多设备同步，是一定要同步线，还是sdk提供软同步的实现？