orbbec/pyorbbecsdk #25: ModuleNotFoundError: No module named 'pyorbbecsdk'

202264642365, 2024-02-01:
eli@jetsonnano:~$ cd ./下载/pyorbbecsdk/
eli@jetsonnano:~/下载/pyorbbecsdk$ python3 -m venv ./venv
eli@jetsonnano:~/下载/pyorbbecsdk$ source venv/bin/activate
(venv) eli@jetsonnano:~/下载/pyorbbecsdk$ python3 examples/depth_viewer.py
Traceback (most recent call last):
  File "examples/depth_viewer.py", line 16, in <module>
    from pyorbbecsdk import Pipeline
ModuleNotFoundError: No module named 'pyorbbecsdk'
(venv) eli@jetsonnano:~/下载/pyorbbecsdk$ export PATHONPATH=$PYTHONPATH:$(pwd)/install/lib/
(venv) eli@jetsonnano:~/下载/pyorbbecsdk$ sudo bash ./scripts/install_udev_rules.sh
[sudo] eli 的密码： 
usb rules file install at /etc/udev/rules.d/99-obsensor-libusb.rules
exit
(venv) eli@jetsonnano:~/下载/pyorbbecsdk$  sudo udevadm control --reload-rules && sudo udevadm trigger
(venv) eli@jetsonnano:~/下载/pyorbbecsdk$  python3 examples/depth_viewer.py
Traceback (most recent call last):
  File "examples/depth_viewer.py", line 16, in <module>
    from pyorbbecsdk import Pipeline
ModuleNotFoundError: No module named 'pyorbbecsdk'
(venv) eli@jetsonnano:~/下载/pyorbbecsdk$ 

按照官网测试sample的代码，反馈找不到pyorbbecsdk模块，这个问题怎么解决？

tanoshiiyo, 2024-02-04:
try to use  export PATHONPATH=$PYTHONPATH:/下载/pyorbbecsdk/install/lib/

apan666555, 2024-09-09:
@wood-kite have you solve it?

Abrahamh08, 2025-03-08:
Exporting PYTHONPATH when installing the module is not a viable solution for me because I use uv which syncs package sources on every run. Please fix this to not rely on PYTHONPATH or to correctly source it during installation