orbbec/pyorbbecsdk #27: fail to run examples/color_view.py and examples/save_image_to_disk.py with Astra S

cg3dland, 2024-02-29:
I follow up the guide to install the Python binding for Orbbec SDK on my Linux PC (Ubuntu 22.04.4), attach an Astra S camera to this Linux PC, and then run the sample app depth_viewer.py successfully, please refer to the attached log depth.log.

But when I run the sample app color_viewer.py, I get following error and app abort abnormally, please refer to the attached log color_view.log:
[02/29 16:11:43.291181][warning][449991][ObException.hpp:40] Start stream failed, OB_SENSOR_COLOR Match openni video mode failed!
[02/29 16:11:43.291247][warning][449991][Pipeline.cpp:233] Execute failure! A libobsensor_exception has occurred!
         - where:233#start
         - msg:Start stream failed, OB_SENSOR_COLOR Match openni video mode failed!
         - type:N11libobsensor12io_exceptionE

BTW，I get following error when I run the sample app save_image_to_disk.py,  and the app abort abnormally, please refer to the attached log save_image.log:
[02/29 16:15:53.070439][warning][450206][ObException.hpp:40] Request failed, statusCode: 6, msg: device response with in
consistent response id, nId=42, nRequestId=43
[02/29 16:15:53.070625][warning][450206][AbstractDevice.cpp:1272] Execute failure! A libobsensor_exception has occurred!         - where:1272#switchD2CParams
         - msg:Request failed, statusCode: 6, msg: device response with inconsistent response id, nId=42, nRequestId=43
         - type:N11libobsensor12io_exceptionE
[02/29 16:15:53.070639][warning][450206][AbstractDevice.cpp:1272] set align hardware mode int failed!
[02/29 16:15:53.070651][warning][450206][OpenNIDevice.cpp:805] current d2c param index out of range!
[02/29 16:15:53.070678][info][450206][Pipeline.cpp:239] Try to start streams!
[02/29 16:15:53.070692][error][450206][OpenNIVideoSensor.cpp:427] Match openni video mode failed!
[02/29 16:15:53.070709][warning][450206][ObException.hpp:40] Start stream failed, OB_SENSOR_COLOR Match openni video mod
e failed!
[02/29 16:15:53.070759][warning][450206][Pipeline.cpp:233] Execute failure! A libobsensor_exception has occurred!
         - where:233#start
         - msg:Start stream failed, OB_SENSOR_COLOR Match openni video mode failed!
         - type:N11libobsensor12io_exceptionE

Could you please help me to resolve the issues and make these 2 apps run at my side with Astra S?

Thanks,

[color_view.log](https://github.com/orbbec/pyorbbecsdk/files/14445488/color_view.log)
[depth.log](https://github.com/orbbec/pyorbbecsdk/files/14445491/depth.log)
[save_image.log](https://github.com/orbbec/pyorbbecsdk/files/14445494/save_image.log)

thaiph99, 2024-04-05:
@cg3dland  I have the same problems, so do you have any solutions for this?
Any maintainer in here?

swarnabha13, 2024-06-14:
@cg3dland I have the exact same issue with Astra S. Were you able to resolve the problem?

Li-Zhi-Hao, 2024-06-28:
Me too, have you solved the problem? Share it

monajalal, 2024-10-18:
same

[10/18 11:34:03.129114][warning][6796][Pipeline.cpp:45] Execute failure! A libobsensor_exception has occurred!
	 - where:45#Pipeline
	 - msg:Current device does not support frame sync!
	 - type:N11libobsensor31unsupported_operation_exceptionE

silenzio777, 2024-12-28:
The same:

"startStream error: Start stream failed, OB_SENSOR_COLOR Match openni video mode failed!"

SDK version: 1.10.16
Camera Name: Astra Pro
Hardware versions: FW=5.8.22 (14), HW=0, Chip=0, Sensor=0, SYS=12
Firmware version: RD107E-007

Apple M2 Pro, MacOSX 13.4.1

Please help!

jujimeizuo, 2025-01-04:
Same as above, I need help!

[image]