orbbec/pyorbbecsdk #31: USB device can not be found

1987604, 2024-03-08:
when I install Python SDK, and try to run ./examples/color_viewer.py, a error is shown. could you please tell me how to do with this problem, thanks
/09 00:44:18.922555][info][4520][Context.cpp:77]     - SDK version: 1.9.5
[03/09 00:44:18.922643][info][4520][Context.cpp:78]     - SDK stage version: main
[03/09 00:44:18.922681][info][4520][Context.cpp:82] get config EnumerateNetDevice:false
[03/09 00:44:18.922768][info][4520][MfPal.cpp:102] createObPal: create WinPal!
[03/09 00:44:18.946211][warning][4520][MfPal.cpp:212] Found a usb port without interface url! drop it!(device url=\\?\USB#VID_2BC5&PID_0402#6&956E9B8&0&2#{A5DCBF10-6530-11D2-901F-00C04FB951ED} interface index=0)
[03/09 00:44:18.946485][info][4520][DeviceManager.cpp:15] Current found device(s): (0)
[03/09 00:44:18.946627][info][4520][Pipeline.cpp:15] Try to create pipeline with default device.
[03/09 00:44:18.946763][warning][4520][ObException.cpp:5] No device found, fail to create pipeline!
Traceback (most recent call last):
  File "D:\works\VS\openniwork\pyorbbecsdk\examples\color_viewer.py", line 67, in <module>
    main()
  File "D:\works\VS\openniwork\pyorbbecsdk\examples\color_viewer.py", line 30, in main
    pipeline = Pipeline()
RuntimeError: Caught an unknown exception!
[03/09 00:44:18.964226][info][4520][Context.cpp:90] Context destroyed

vladimir-metralight, 2024-04-04:
Exactly same problem: pyorbbec sdk examples does not work, but prebuilt *.exe examples work.

x18-1, 2024-07-05:
Did you install the camera driver in their oficial website?

wwzee-hub, 2025-01-08:
> 完全相同的问题：pyorbbec sdk 示例不起作用，但预构建的 *.exe 示例有效。

bro,do you deal with it ,can you give me a help