orbbec/pyorbbecsdk #38: 无法打开深度相机获取数据

ys1215, 2024-05-15:
[05/15 23:53:05.285380][info][3844][Context.cpp:68] Context created with config: default config!
[05/15 23:53:05.285398][info][3844][Context.cpp:73] Work directory=D:\YS\project\code\4kinectdk , SDK version=v1.10.5-20240511-7dc8be970
[05/15 23:53:05.285427][info][3844][MfPal.cpp:102] createObPal: create WinPal!
[05/15 23:53:05.327431][info][3844][DeviceManager.cpp:15] Current found device(s): (1)
[05/15 23:53:05.327442][info][3844][DeviceManager.cpp:24] 	- Name: Femto Bolt, PID: 0x066b, SN/ID: CL8R24100E0, Connection: USB3.1
[05/15 23:53:05.327482][info][3844][Pipeline.cpp:15] Try to create pipeline with default device.
[05/15 23:53:05.327530][info][3844][FemtoBoltUvcDevice.cpp:23] FemtoBoltUvcDevice init ...
[05/15 23:53:05.327659][info][3844][FemtoBoltUvcDevice.cpp:121] Create command start!
[05/15 23:53:05.327669][info][3844][MfPal.cpp:289] Create MSDEConverterDevice uvc device.
[05/15 23:53:05.330790][error][3844][MSDEConverterDevice.cpp:723] Failed to load depth engine plugin
[05/15 23:53:06.171562][info][3844][AbstractDevice.cpp:117] 	- Firmware version: 1.0.9
[05/15 23:53:06.171574][info][3844][FemtoBoltUvcDevice.cpp:280] Create command done!
[05/15 23:53:06.171580][info][3844][FemtoBoltUvcDevice.cpp:401] init sensor map start!
[05/15 23:53:06.171596][info][3844][FemtoBoltUvcDevice.cpp:428] init sensor map done!
[05/15 23:53:06.171683][info][3844][FemtoBoltUvcDevice.cpp:284] Init depth process param start!
[05/15 23:53:06.530716][info][3844][MSDEConverterDevice.cpp:777] got nvram data succeed.
[05/15 23:53:07.086397][info][3844][FemtoBoltUvcDevice.cpp:358] setNvramDataStreamStopFunc succeed
[05/15 23:53:07.086630][info][3844][FemtoBoltUvcDevice.cpp:397] Init depth process param done!
[05/15 23:53:07.087961][info][3844][FemtoBoltUvcDevice.cpp:38] FemtoBoltUvcDevice init done!
[05/15 23:53:07.087979][info][3844][DeviceManager.cpp:150] Device created successfully! Name: Femto Bolt, PID: 0x066b, SN/ID: CL8R24100E0
[05/15 23:53:07.088020][info][3844][Pipeline.cpp:47] Pipeline created with device: {name: Femto Bolt, sn: CL8R24100E0}, @0x18BB27A05B0
[05/15 23:53:07.088951][info][3844][FemtoBoltUvcDevice.cpp:528] Depth sensor has been created!
[05/15 23:53:07.089006][info][3844][VideoSensor.cpp:382] Query stream profile! size= 14, SensorType=OB_SENSOR_DEPTH
[05/15 23:53:07.089019][info][3844][VideoSensor.cpp:384]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 640, height: 576, fps: 15}
[05/15 23:53:07.089028][info][3844][VideoSensor.cpp:384]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 1024, height: 1024, fps: 15}
[05/15 23:53:07.089040][info][3844][VideoSensor.cpp:384]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 1024, height: 1024, fps: 5}
[05/15 23:53:07.089050][info][3844][VideoSensor.cpp:384]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 640, height: 576, fps: 30}
[05/15 23:53:07.089058][info][3844][VideoSensor.cpp:384]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 320, height: 288, fps: 30}
[05/15 23:53:07.089066][info][3844][VideoSensor.cpp:384]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 640, height: 576, fps: 25}
[05/15 23:53:07.089073][info][3844][VideoSensor.cpp:384]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 320, height: 288, fps: 25}
[05/15 23:53:07.089081][info][3844][VideoSensor.cpp:384]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 320, height: 288, fps: 15}
[05/15 23:53:07.089089][info][3844][VideoSensor.cpp:384]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 640, height: 576, fps: 5}
[05/15 23:53:07.089096][info][3844][VideoSensor.cpp:384]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 320, height: 288, fps: 5}
[05/15 23:53:07.089104][info][3844][VideoSensor.cpp:384]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 512, height: 512, fps: 30}
[05/15 23:53:07.089112][info][3844][VideoSensor.cpp:384]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 512, height: 512, fps: 25}
[05/15 23:53:07.089119][info][3844][VideoSensor.cpp:384]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 512, height: 512, fps: 15}
[05/15 23:53:07.089127][info][3844][VideoSensor.cpp:384]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 512, height: 512, fps: 5}
depth profile:  <VideoStreamProfile: 640x576@15>
[05/15 23:53:07.089984][info][3844][Pipeline.cpp:237] Try to start streams!
[05/15 23:53:07.090061][info][3844][VideoSensor.cpp:631] start OB_SENSOR_DEPTH stream with profile: {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 640, height: 576, fps: 15}, Backend Format: OB_FORMAT_Y16
[05/15 23:53:07.090071][info][3844][MSDEConverterDevice.cpp:549] Start real profile,width:7680 height:434
[05/15 23:53:07.093510][info][6304][MSDEConverterDevice.cpp:67] Depth engine got nvram data size:492464
[05/15 23:53:07.093536][error][6304][MSDEConverterDevice.cpp:91] Failed to load depth engine plugin,init depth engine failed.
[05/15 23:53:07.395049][info][6304][MSDEConverterDevice.cpp:67] Depth engine got nvram data size:492464
[05/15 23:53:07.395092][error][6304][MSDEConverterDevice.cpp:91] Failed to load depth engine plugin,init depth engine failed.
[05/15 23:53:07.706056][info][6304][MSDEConverterDevice.cpp:67] Depth engine got nvram data size:492464
[05/15 23:53:07.706101][error][6304][MSDEConverterDevice.cpp:91] Failed to load depth engine plugin,init depth engine failed.
[05/15 23:53:08.017962][info][6304][MSDEConverterDevice.cpp:67] Depth engine got nvram data size:492464
[05/15 23:53:08.018035][error][6304][MSDEConverterDevice.cpp:91] Failed to load depth engine plugin,init depth engine failed.
[05/15 23:53:08.329468][info][6304][MSDEConverterDevice.cpp:67] Depth engine got nvram data size:492464
[05/15 23:53:08.329517][error][6304][MSDEConverterDevice.cpp:91] Failed to load depth engine plugin,init depth engine failed.
[05/15 23:53:08.639933][info][6304][MSDEConverterDevice.cpp:67] Depth engine got nvram data size:492464
[05/15 23:53:08.639982][error][6304][MSDEConverterDevice.cpp:91] Failed to load depth engine plugin,init depth engine failed.
[05/15 23:53:08.952420][info][6304][MSDEConverterDevice.cpp:67] Depth engine got nvram data size:492464
[05/15 23:53:08.952479][error][6304][MSDEConverterDevice.cpp:91] Failed to load depth engine plugin,init depth engine failed.
[05/15 23:53:09.262126][info][6304][MSDEConverterDevice.cpp:67] Depth engine got nvram data size:492464
[05/15 23:53:09.262150][error][6304][MSDEConverterDevice.cpp:91] Failed to load depth engine plugin,init depth engine failed.
[05/15 23:53:09.574891][info][6304][MSDEConverterDevice.cpp:67] Depth engine got nvram data size:492464
[05/15 23:53:09.574988][error][6304][MSDEConverterDevice.cpp:91] Failed to load depth engine plugin,init depth engine failed.
[05/15 23:53:09.882068][info][6304][MSDEConverterDevice.cpp:67] Depth engine got nvram data size:492464
[05/15 23:53:09.882187][error][6304][MSDEConverterDevice.cpp:91] Failed to load depth engine plugin,init depth engine failed.
[05/15 23:53:10.190269][info][3844][Pipeline.cpp:250] Start streams done!
[05/15 23:53:10.190282][info][3844][Pipeline.cpp:226] Pipeline start done!

---------------------------------------------------------------
如上是报错信息，不知道如何解决。 看到网上有说是电脑GPU的问题，不知道官方是否有准确的解答和解决方案。不胜感激~

jian-dong, 2024-05-16:
需要把install后的 `depthengine_2_0.dll`拷贝到运行目录

TeleeMa, 2024-05-30:
我也不行，我是dpkg安装的，请问您是怎么解决的 @jian-dong @ys1215

elijahatdsi, 2025-03-04:
# Solution for Raspberry Pi 4

While you do need to make sure to have the dll copied into the running directory, the depth engine is missing a dependancy when you first set it up so you need to install it.

Check depth engine dependancies:

```bash
ldd libdepthengine.so
```

Install   libegl1-mesa
```bash
 sudo apt-get install libegl1-mesa
```

This worked for me after finding no solutions on any of the forums lol...hopefully it can help you from pulling your hair out as well.