orbbec/pyorbbecsdk #44: Femto Bolt cannot acquire depth-view images in docker

TeleeMa, 2024-06-04:
Hi,

Femto Bolt can only get color images, but cannot get depth-view images in docker. However, when out of docker, both the rgb and depth view work normally.

The error is shown as 
[image]

Thanks.

jian-dong, 2024-06-05:
Hi,

Femto Bolt relies on NVIDIA's GPU. To run it inside Docker, you need to install the NVIDIA Docker toolkit. The installation guide is available here: https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/latest/install-guide.html.

After installation, run `nvidia-smi` inside the Docker container to check if the GPU is available.

Best regards,

samueltorche, 2024-06-25:
Hey, 

I installed the NVIDIA Docker toolkit and tested it successfully using their sample workload in Docker https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/latest/sample-workload.html# where my GPU shows up in the output.

However, the problem still persists with the same errors and warnings as the screenshot above.
@TeleeMa Did you manage to get the depth image in Docker ?

Kind regards

tushardmaske, 2024-08-01:
Hey, 

I am facing the same issue. Could you find any solution ?

Best Regards

samueltorche, 2024-08-06:
Hello,

Nope I did not find any solution to this problem, I hope it gets fixed though.