orbbec/pyorbbecsdk #45: 测试样例depth_viewer.py无法成功启动或成功连接几秒后断开

zzy257248, 2024-06-07:
您好，我使用的是Femto Bolt相机，环境是Ubuntu20.04，按教程搭建好venv环境后运行python3 examples/depth_viewer.py
有时无法成功启动，需要重新插拔USB线，报错如下：
`
[06/07 17:23:17.985307][info][240308][Context.cpp:68] Context created with config: /home/xxx/pyorbbecsdk/config/OrbbecSDKConfig_v1.0.xml
[06/07 17:23:17.985405][info][240308][Context.cpp:73] Work directory=/home/xxx/pyorbbecsdk, SDK version=v1.10.8-20240604-d1671a3
[06/07 17:23:17.985500][info][240308][LinuxPal.cpp:32] createObPal: create LinuxPal!
[06/07 17:23:21.465093][info][240308][LinuxPal.cpp:143] Create PollingDeviceWatcher!
[06/07 17:23:21.465154][info][240308][DeviceManager.cpp:15] Current found device(s): (1)
[06/07 17:23:21.465168][info][240308][DeviceManager.cpp:24] 	- Name: Femto Bolt, PID: 0x066b, SN/ID: , Connection: USB3.1
[06/07 17:23:21.465236][info][240308][Pipeline.cpp:15] Try to create pipeline with default device.
[06/07 17:23:21.465315][info][240308][FemtoBoltUvcDevice.cpp:23] FemtoBoltUvcDevice init ...
[06/07 17:23:21.465473][info][240308][FemtoBoltUvcDevice.cpp:121] Create command start!
[06/07 17:23:21.465523][error][240308][DeviceLibusb.cpp:112] failed to open usb device!  error: OB_USB_STATUS_IO
[06/07 17:23:21.465704][warning][240308][EnumeratorLibusb.cpp:341] Execute failure! A std::exception has occurred!
	 - where:341#createUsbDevice
	 - msg:failed to open usb device!  error: OB_USB_STATUS_IO
	 - type:St13runtime_error
[06/07 17:23:21.465723][warning][240308][EnumeratorLibusb.cpp:341] failed to create usb device at index: 3, url:2-4-69
[06/07 17:23:21.966036][error][240308][DeviceLibusb.cpp:112] failed to open usb device!  error: OB_USB_STATUS_IO
[06/07 17:23:21.966192][warning][240308][EnumeratorLibusb.cpp:341] Execute failure! A std::exception has occurred!
	 - where:341#createUsbDevice
	 - msg:failed to open usb device!  error: OB_USB_STATUS_IO
	 - type:St13runtime_error
[06/07 17:23:21.966322][warning][240308][EnumeratorLibusb.cpp:341] failed to create usb device at index: 3, url:2-4-69
[06/07 17:23:21.966369][warning][240308][ObException.hpp:40] usbEnumerator create UnpackDevicePort failed!
Traceback (most recent call last):
  File "examples/depth_viewer.py", line 104, in <module>
    main()
  File "examples/depth_viewer.py", line 47, in main
    pipeline = Pipeline()
RuntimeError: Caught an unknown exception!
[06/07 17:23:21.978441][info][240308][Context.cpp:84] Context destroyed
`
有时可以成功输出实时图像但持续几秒后就会卡住并断开连接，报错如下：
`
[06/07 17:23:32.076732][info][240420][Context.cpp:68] Context created with config: /home/xxx/pyorbbecsdk/config/OrbbecSDKConfig_v1.0.xml
[06/07 17:23:32.076744][info][240420][Context.cpp:73] Work directory=/home/xxx/pyorbbecsdk, SDK version=v1.10.8-20240604-d1671a3
[06/07 17:23:32.076754][info][240420][LinuxPal.cpp:32] createObPal: create LinuxPal!
[06/07 17:23:32.078030][info][240420][LinuxPal.cpp:143] Create PollingDeviceWatcher!
[06/07 17:23:32.078045][info][240420][DeviceManager.cpp:15] Current found device(s): (1)
[06/07 17:23:32.078048][info][240420][DeviceManager.cpp:24] 	- Name: Femto Bolt, PID: 0x066b, SN/ID: CL8FC31009N, Connection: USB3.1
[06/07 17:23:32.078061][info][240420][Pipeline.cpp:15] Try to create pipeline with default device.
[06/07 17:23:32.078082][info][240420][FemtoBoltUvcDevice.cpp:23] FemtoBoltUvcDevice init ...
[06/07 17:23:32.078127][info][240420][FemtoBoltUvcDevice.cpp:121] Create command start!
[06/07 17:23:32.078924][info][240420][MSDEConverterDevice.cpp:726] Succeed to load depth engine plugin
[06/07 17:23:32.744287][info][240420][AbstractDevice.cpp:117] 	- Firmware version: 1.0.9
[06/07 17:23:32.744321][info][240420][FemtoBoltUvcDevice.cpp:280] Create command done!
[06/07 17:23:32.744332][info][240420][FemtoBoltUvcDevice.cpp:401] init sensor map start!
[06/07 17:23:32.744370][info][240420][FemtoBoltUvcDevice.cpp:428] init sensor map done!
[06/07 17:23:32.744632][info][240420][FemtoBoltUvcDevice.cpp:284] Init depth process param start!
[06/07 17:23:33.088138][info][240420][MSDEConverterDevice.cpp:777] got nvram data succeed.
[06/07 17:23:33.352296][info][240420][FemtoBoltUvcDevice.cpp:358] setNvramDataStreamStopFunc succeed
[06/07 17:23:33.352406][info][240420][FemtoBoltUvcDevice.cpp:397] Init depth process param done!
[06/07 17:23:33.353901][info][240420][FemtoBoltUvcDevice.cpp:38] FemtoBoltUvcDevice init done!
[06/07 17:23:33.353956][info][240420][DeviceManager.cpp:150] Device created successfully! Name: Femto Bolt, PID: 0x066b, SN/ID: CL8FC31009N
[06/07 17:23:33.354047][info][240420][Pipeline.cpp:47] Pipeline created with device: {name: Femto Bolt, sn: CL8FC31009N}, @0x1F097A0
[06/07 17:23:33.354989][info][240420][FemtoBoltUvcDevice.cpp:528] Depth sensor has been created!
[06/07 17:23:33.355137][info][240420][VideoSensor.cpp:384] Query stream profile! size= 14, SensorType=OB_SENSOR_DEPTH
[06/07 17:23:33.355160][info][240420][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 640, height: 576, fps: 15}
[06/07 17:23:33.355175][info][240420][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 1024, height: 1024, fps: 15}
[06/07 17:23:33.355189][info][240420][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 1024, height: 1024, fps: 5}
[06/07 17:23:33.355203][info][240420][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 640, height: 576, fps: 30}
[06/07 17:23:33.355216][info][240420][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 320, height: 288, fps: 30}
[06/07 17:23:33.355232][info][240420][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 640, height: 576, fps: 25}
[06/07 17:23:33.355246][info][240420][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 320, height: 288, fps: 25}
[06/07 17:23:33.355259][info][240420][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 320, height: 288, fps: 15}
[06/07 17:23:33.355273][info][240420][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 640, height: 576, fps: 5}
[06/07 17:23:33.355286][info][240420][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 320, height: 288, fps: 5}
[06/07 17:23:33.355319][info][240420][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 512, height: 512, fps: 30}
[06/07 17:23:33.355335][info][240420][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 512, height: 512, fps: 25}
[... 35 lines cut ...]
[06/07 17:23:42.003906][info][240470][DeviceManager.cpp:15] Current device(s) list: (0)
[06/07 17:23:42.677732][warning][240420][Pipeline.cpp:377] Wait for frame timeout, you can try to increase the wait time! current timeout=100 [**9 logs in 6024ms**]
[06/07 17:23:48.414077][info][240530][DeviceManager.cpp:181] Device changed! removed: 0, added: 1
[06/07 17:23:48.414164][info][240530][DeviceManager.cpp:15] Current device(s) list: (1)
[06/07 17:23:48.414205][info][240530][DeviceManager.cpp:24] 	- Name: Femto Bolt, PID: 0x066b, SN/ID: , Connection: USB3.1
[06/07 17:23:49.003172][info][240532][DeviceManager.cpp:181] Device changed! removed: 1, added: 0
[06/07 17:23:49.003245][info][240532][DeviceManager.cpp:15] Removed device(s) list: (1)
[06/07 17:23:49.003266][info][240532][DeviceManager.cpp:24] 	- Name: Femto Bolt, PID: 0x066b, SN/ID: , Connection: USB3.1
[06/07 17:23:49.003286][info][240532][DeviceManager.cpp:15] Current device(s) list: (0)
[06/07 17:23:54.697205][warning][240420][Pipeline.cpp:377] Wait for frame timeout, you can try to increase the wait time! current timeout=100 [**120 logs in 12019ms**]
[06/07 17:23:56.738653][info][240584][DeviceManager.cpp:181] Device changed! removed: 0, added: 1
[06/07 17:23:56.738754][info][240584][DeviceManager.cpp:15] Current device(s) list: (1)
[06/07 17:23:56.738772][info][240584][DeviceManager.cpp:24] 	- Name: Femto Bolt, PID: 0x066b, SN/ID: , Connection: USB3.1
[06/07 17:23:57.049883][info][240585][DeviceManager.cpp:181] Device changed! removed: 1, added: 0
[06/07 17:23:57.049899][info][240585][DeviceManager.cpp:15] Removed device(s) list: (1)
[06/07 17:23:57.049904][info][240585][DeviceManager.cpp:24] 	- Name: Femto Bolt, PID: 0x066b, SN/ID: , Connection: USB3.1
[06/07 17:23:57.049910][info][240585][DeviceManager.cpp:15] Current device(s) list: (0)
[06/07 17:23:58.927189][info][240617][DeviceManager.cpp:181] Device changed! removed: 0, added: 1
[... cut ...]
[06/07 17:24:18.736048][warning][240420][Pipeline.cpp:377] Wait for frame timeout, you can try to increase the wait time! current timeout=100 [**240 logs in 24038ms**]
[06/07 17:25:06.815638][warning][240420][Pipeline.cpp:377] Wait for frame timeout, you can try to increase the wait time! current timeout=100 [**480 logs in 48079ms**]
[06/07 17:26:06.914316][warning][240420][Pipeline.cpp:377] Wait for frame timeout, you can try to increase the wait time! current timeout=100 [**600 logs in 60098ms**]
`
事实上，我在运行OrbbecViewer或拿相机运行手眼标定程序时也常常遇到成功运行几秒后就卡住断连的问题，相机背后的指示灯是常亮的，应该不是硬件的问题。

jian-dong, 2024-06-07:
@zhonghong322

zhonghong322, 2024-06-11:
从Log看，设备出现了断开。1、Femto Bolt 你是用12v 2A的适配器供电的吗。2、USB 线是直连到PC，还是经过Hub。

zzy257248, 2024-06-13:
您好
1. 我的相机是通过原装适配器加一根DC延长线供电的，但去掉DC延长线直接供电后该问题依然存在；
2. 我的USB线是通过原装数据线加一根USB延长线直连到PC的USB3.0接口的。

zhonghong322, 2024-06-14:
1、麻烦使用OrbbecViewer 看下你的Bolt的固件版本，还有你的PC的配置是怎样的。  2、使用OrbbecViewer 开流会断开吗。

zhonghong322, 2024-06-14:
你的固件版本是什么?

zzy257248, 2024-06-14:
固件版本1.0.9，SDK版本1.8.3

zhonghong322, 2024-06-21:
1、你在windows上测试过吗，有没有同样的问题。2. 或者下载下最新的Orbbec Viewer v1.10.8测试下。

xcy2011sky, 2024-06-21:
怀疑是usb链接不够稳定。可以试下，这个[usb延长线](https://item.m.jd.com/product/100010433025.html?utm_term=CopyURL&utm_user=plusmember&utm_source=iosapp&utm_campaign=t_335139774&utm_medium=appshare&ad_od=share&gx=RnAomTM2G1CFo_B82dIsF-t73cM8NQ&gxd=RnAowzVePGeImM0Rp4d2XLwCOGMqcX-gi_yTcSXnEtGWPd1VYBRQRnIhMKx8HjA)线材。