orbbec/pyorbbecsdk #47: Breaks with numpy >= 2.0

lkskstlr, 2024-06-19:
Hi Team,

Thanks for the amazing camera and SDK!

The examples don't work if the environment has numpy>=2.0 and the errors are very subtle. It would probably be good to guard on this during the install.

Cheers
Lukas

```shell
conda create -n test python=3.10.12
conda activate test
git clone --branch v1.3.2 --depth 1 https://github.com/orbbec/pyorbbecsdk.git
cd pyorbbecsdk
pip install -r requirements.txt
cmake -DPython3_ROOT_DIR="$CONDA_PPREFIX" -Dpybind11_DIR=`pybind11-config --cmakedir` ..
make -j
make install
cd ..
export PYTHONPATH=$PYTHONPATH:$(pwd)/install/lib/
python examples/net_device.py
[06/19 11:04:34.404272][info][3419649][Context.cpp:68] Context created with config: /home/lukas/playground/pyorbbecsdk/config/OrbbecSDKConfig_v1.0.xml
[06/19 11:04:34.404343][info][3419649][Context.cpp:73] Work directory=/home/lukas/playground/pyorbbecsdk, SDK version=v1.10.5-20240511-7dc8be9
[06/19 11:04:34.404376][info][3419649][LinuxPal.cpp:31] createObPal: create LinuxPal!
[06/19 11:04:34.404665][info][3419649][LinuxPal.cpp:142] Create PollingDeviceWatcher!
[06/19 11:04:34.404701][info][3419649][DeviceManager.cpp:15] Current found device(s): (0)
Enter the ip address of the device (default: 192.168.1.10): 10.157.29.60

[06/19 11:04:39.731241][info][3419649][AbstractDevice.cpp:117] 	- Firmware version: 1.2.7
[06/19 11:04:39.819302][info][3419649][FemtoMegaNetDevice.cpp:132] FemtoMega Net device created! PID: 0x0669, SN: CL2N24100CE
[06/19 11:04:39.819315][info][3419649][DeviceManager.cpp:99] create Net Device success! address=10.157.29.60, port=8090
[06/19 11:04:39.819343][info][3419649][Pipeline.cpp:47] Pipeline created with device: {name: Femto Mega, sn: CL2N24100CE}, @0x2790480
[06/19 11:04:39.819556][info][3419649][FemtoMegaNetDevice.cpp:444] Color sensor has been created!
[06/19 11:04:39.819583][warning][3419649][ObException.cpp:5] Invalid input, No matched video stream profile found!
[06/19 11:04:39.819757][info][3419649][FemtoMegaNetDevice.cpp:394] Depth sensor has been created!
[06/19 11:04:39.819766][warning][3419649][ObException.cpp:5] Invalid input, No matched video stream profile found!
[06/19 11:04:39.820411][info][3419649][Pipeline.cpp:237] Try to start streams!
[06/19 11:04:39.820435][info][3419649][VideoSensor.cpp:631] start OB_SENSOR_COLOR stream with profile: {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1920, height: 1080, fps: 15}, Backend Format: OB_FORMAT_MJPG
[06/19 11:04:40.553509][info][3419649][VideoSensor.cpp:631] start OB_SENSOR_DEPTH stream with profile: {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 640, height: 576, fps: 15}, Backend Format: OB_FORMAT_Y16
[06/19 11:04:41.433001][info][3419649][Pipeline.cpp:250] Start streams done!
[06/19 11:04:41.433009][info][3419649][Pipeline.cpp:226] Pipeline start done!
[06/19 11:04:41.444249][info][3419649][Pipeline.cpp:360] Try to stop pipeline!
[06/19 11:04:41.444266][info][3419649][Pipeline.cpp:332] Try to stop streams!
[06/19 11:04:41.600140][info][3419649][VideoSensor.cpp:719] Video sensor stopped, @OB_SENSOR_DEPTH
[06/19 11:04:41.600153][info][3419649][Pipeline.cpp:345] Sensor stream stopped, sensorType=OB_SENSOR_DEPTH
[06/19 11:04:41.816692][info][3419649][VideoSensor.cpp:719] Video sensor stopped, @OB_SENSOR_COLOR
[06/19 11:04:41.816700][info][3419649][Pipeline.cpp:345] Sensor stream stopped, sensorType=OB_SENSOR_COLOR
[06/19 11:04:41.816702][info][3419649][Pipeline.cpp:351] Stop streams done!
[06/19 11:04:41.817092][info][3419649][Pipeline.cpp:389] Stop pipeline done!
Traceback (most recent call last):
  File "/home/lukas/playground/pyorbbecsdk/examples/net_device.py", line 123, in <module>
    main()
  File "/home/lukas/playground/pyorbbecsdk/examples/net_device.py", line 97, in main
    depth_data = np.frombuffer(depth_frame.get_data(), dtype=np.uint16).reshape(depth_frame.get_height(),
ValueError: ndarray is not C-contiguous
[06/19 11:04:41.819611][info][3419649][Pipeline.cpp:360] Try to stop pipeline!
[06/19 11:04:41.819625][info][3419649][Pipeline.cpp:332] Try to stop streams!
[06/19 11:04:41.819635][warning][3419649][VideoSensor.cpp:665] Sensor(OB_SENSOR_DEPTH) already stop.
[06/19 11:04:41.819638][info][3419649][Pipeline.cpp:345] Sensor stream stopped, sensorType=OB_SENSOR_DEPTH
[06/19 11:04:41.819643][warning][3419649][VideoSensor.cpp:665] Sensor(OB_SENSOR_COLOR) already stop.
[06/19 11:04:41.819645][info][3419649][Pipeline.cpp:345] Sensor stream stopped, sensorType=OB_SENSOR_COLOR
[06/19 11:04:41.819648][info][3419649][Pipeline.cpp:351] Stop streams done!
[06/19 11:04:41.820260][info][3419649][Pipeline.cpp:389] Stop pipeline done!
[06/19 11:04:41.820274][info][3419649][Pipeline.cpp:75] Pipeline destroyed! @0x2790480
[06/19 11:04:41.820279][info][3419649][FemtoMegaNetDevice.cpp:136] FemtoMega Net device destroyed! PID: 0x0669, SN: CL2N24100CE
[06/19 11:04:41.820287][info][3419649][VideoSensor.cpp:301] VideoSensor destroyed, @OB_SENSOR_COLOR
[06/19 11:04:41.820295][info][3419649][VideoSensor.cpp:301] VideoSensor destroyed, @OB_SENSOR_DEPTH
[06/19 11:04:41.820313][warning][3419649][RTSPStreamPort.cpp:294] Stream have not been started!
[06/19 11:04:41.820552][warning][3419649][RTSPStreamPort.cpp:294] Stream have not been started!
[06/19 11:04:42.821499][info][3419649][Context.cpp:87] Context destroyed
[... 55 lines cut ...]

CMake Output
```
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Host platform set to: linux_x64
-- Found Python3: /home/lukas/miniconda3/envs/test/bin/python3.10 (found version "3.10.12") found components: Interpreter Development Development.Module Development.Embed 
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- Found pybind11: /home/lukas/miniconda3/envs/test/lib/python3.10/site-packages/pybind11/include (found version "2.11.0")
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lukas/playground/pyorbbecsdk/build

```

closed 2024-06-21 by jian-dong

monajalal, 2024-10-14:
@lkskstlr 
do you know why I still get an error using your fix? https://github.com/orbbec/pyorbbecsdk/issues/79

```
Original error was: No module named 'numpy.core._multiarray_umath'

(venv) (base) 2024-10-14 12:02:43 [mona@DOS pyorbbecsdk]$ pip install "numpy<2.0"
Looking in indexes: https://pypi.org/simple, https://pypi.ngc.nvidia.com
Requirement already satisfied: numpy<2.0 in /home/mona/anaconda3/lib/python3.9/site-packages (1.23.5)

```

lintianfang, 2025-04-29:
Do you know how to fix this issue?