orbbec/pyorbbecsdk #49: 无法寻找到设备

x18-1, 2024-07-03:
当我去运行hello_orbbec.py，报错如下：
```python
PS D:\code\CV> & d:/code/CV/env/Scripts/python.exe d:/code/CV/pyorbbecsdk/examples/hello_orbbec.py
Hello Orbbec!
SDK version: 1.10.8
[07/03 17:01:33.000105][info][20452][Context.cpp:68] Context created with config: default config!
[07/03 17:01:33.000466][info][20452][Context.cpp:73] Work directory=D:\code\CV, SDK version=v1.10.8-20240604-d1671a341
[07/03 17:01:33.000876][info][20452][MfPal.cpp:103] createObPal: create WinPal!
[07/03 17:01:35.185784][info][20452][DeviceManager.cpp:15] Current found device(s): (0)
No device connected
[07/03 17:01:35.309079][info][20452][Context.cpp:84] Context destroyed
```
问题出现在这里：
```python
device_list = context.query_devices()

```
device_list的结果总是为0

x18-1, 2024-07-03:
解决了，最终发现因为驱动未安装。希望在文档中提一下。

closed 2024-07-03 by x18-1

LeehanLee, 2025-01-07:
> 解决了，最终发现因为驱动未安装。希望在文档中提一下。

请问下具体是什么驱动没有安装？

x18-1, 2025-01-07:
@LeehanLee 这个相机的驱动，得去官网下载

LeehanLee, 2025-01-07:
能不能好心给个链接，我在奥比中光的官网翻了一天了看到的都不太像

x18-1, 2025-01-07:
@LeehanLee https://vcp.developer.orbbec.com.cn/resourceCenter?defaultSelectedKeys=45

你看看这个行不行

LeehanLee, 2025-01-07:
> @LeehanLee https://vcp.developer.orbbec.com.cn/resourceCenter?defaultSelectedKeys=45
> 
> 你看看这个行不行

就是这个，确实可以了，感谢！！