orbbec/pyorbbecsdk #50: 跑 python pyorbbecsdk/examples/align_filter_viewer.py 。报错。尝试增加等待时间，则仍然报错  段错误 (核心已转储)

biosbeta, 2024-07-05:
[07/05 11:46:35.330759][info][225130][Context.cpp:68] Context created with config: default config!
[07/05 11:46:35.330770][info][225130][Context.cpp:73] Work directory=/home/bios/catkin_franka1, SDK version=v1.10.8-20240604-d1671a3
[07/05 11:46:35.330780][info][225130][LinuxPal.cpp:32] createObPal: create LinuxPal!
[07/05 11:46:35.543548][info][225130][LinuxPal.cpp:143] Create PollingDeviceWatcher!
[07/05 11:46:35.543567][info][225130][DeviceManager.cpp:15] Current found device(s): (1)
[07/05 11:46:35.543597][info][225130][DeviceManager.cpp:24]     - Name: Femto Mega, PID: 0x0669, SN/ID: CL3BA31001N, Connection: USB3.2
[07/05 11:46:35.543603][info][225130][Pipeline.cpp:15] Try to create pipeline with default device.
[07/05 11:46:36.303287][info][225130][AbstractDevice.cpp:117]   - Firmware version: 1.2.7
[07/05 11:46:36.919854][info][225130][FemtoMegaUvcDevice.cpp:49] FemtoMega UVC device created! PID: 0x0669, SN: CL3BA31001N
[07/05 11:46:36.919876][info][225130][DeviceManager.cpp:150] Device created successfully! Name: Femto Mega, PID: 0x0669, SN/ID: CL3BA31001N
[07/05 11:46:36.919907][info][225130][Pipeline.cpp:47] Pipeline created with device: {name: Femto Mega, sn: CL3BA31001N}, @0x1694860
[07/05 11:46:36.921958][info][225130][FemtoMegaUvcDevice.cpp:383] Color sensor has been created!
[07/05 11:46:36.922158][info][225130][VideoSensor.cpp:384] Query stream profile! size= 108, SensorType=OB_SENSOR_COLOR
[07/05 11:46:36.922173][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1920, height: 1080, fps: 15}
[07/05 11:46:36.922181][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 720, fps: 30}
[07/05 11:46:36.922188][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 720, fps: 25}
[07/05 11:46:36.922195][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 720, fps: 15}
[07/05 11:46:36.922201][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 720, fps: 5}
[07/05 11:46:36.922207][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 960, fps: 30}
[07/05 11:46:36.922215][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 960, fps: 25}
[07/05 11:46:36.922221][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 960, fps: 15}
[07/05 11:46:36.922228][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 960, fps: 5}
[07/05 11:46:36.922237][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1920, height: 1080, fps: 30}
[07/05 11:46:36.922268][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1920, height: 1080, fps: 25}
[07/05 11:46:36.922275][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1920, height: 1080, fps: 15}
[07/05 11:46:36.922282][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1920, height: 1080, fps: 5}
[07/05 11:46:36.922288][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 2560, height: 1440, fps: 25}
[07/05 11:46:36.922302][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 2560, height: 1440, fps: 15}
[07/05 11:46:36.922309][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 2560, height: 1440, fps: 5}
[07/05 11:46:36.922315][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 3840, height: 2160, fps: 25}
[07/05 11:46:36.922321][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 3840, height: 2160, fps: 15}
[07/05 11:46:36.922327][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 3840, height: 2160, fps: 5}
[07/05 11:46:36.922333][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 720, fps: 30}
[07/05 11:46:36.922339][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 720, fps: 25}
[07/05 11:46:36.922345][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 720, fps: 15}
[07/05 11:46:36.922351][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 720, fps: 5}
[07/05 11:46:36.922357][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 960, fps: 30}
[07/05 11:46:36.922363][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 960, fps: 25}
[07/05 11:46:36.922370][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 960, fps: 15}
[07/05 11:46:36.922376][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 960, fps: 5}
[07/05 11:46:36.922382][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1920, height: 1080, fps: 30}
[07/05 11:46:36.922388][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1920, height: 1080, fps: 25}
[07/05 11:46:36.922394][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1920, height: 1080, fps: 5}
[07/05 11:46:36.922400][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 2560, height: 1440, fps: 25}
[07/05 11:46:36.922406][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 2560, height: 1440, fps: 15}
[07/05 11:46:36.922412][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 2560, height: 1440, fps: 5}
[07/05 11:46:36.922418][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 3840, height: 2160, fps: 25}
[07/05 11:46:36.922424][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 3840, height: 2160, fps: 15}
[07/05 11:46:36.922430][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 3840, height: 2160, fps: 5}
[07/05 11:46:36.922436][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_H264, width: 1280, height: 720, fps: 30}
[07/05 11:46:36.922443][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_H264, width: 1280, height: 720, fps: 25}
[07/05 11:46:36.922449][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_H264, width: 1280, height: 720, fps: 15}
[07/05 11:46:36.922455][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_H264, width: 1280, height: 720, fps: 5}
[07/05 11:46:36.922461][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_H264, width: 1280, height: 960, fps: 30}
[07/05 11:46:36.922468][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_H264, width: 1280, height: 960, fps: 25}
[07/05 11:46:36.922480][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_H264, width: 1280, height: 960, fps: 15}
[07/05 11:46:36.922487][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_H264, width: 1280, height: 960, fps: 5}
[07/05 11:46:36.922493][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_H264, width: 1920, height: 1080, fps: 30}
[07/05 11:46:36.922500][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_H264, width: 1920, height: 1080, fps: 25}
[07/05 11:46:36.922506][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_H264, width: 1920, height: 1080, fps: 15}
[07/05 11:46:36.922513][info][225130][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_H264, width: 1920, height: 1080, fps: 5}
[... cut ...]
[07/05 11:46:36.934440][info][225130][VideoSensor.cpp:646] start OB_SENSOR_DEPTH stream with profile: {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 640, height: 576, fps: 15}, Backend Format: OB_FORMAT_Y16
[07/05 11:46:37.256192][info][225130][Pipeline.cpp:301] Start streams done!
[07/05 11:46:37.256207][info][225130][Pipeline.cpp:277] Pipeline start done!
[07/05 11:46:37.356632][warning][225130][Pipeline.cpp:377] Wait for frame timeout, you can try to increase the wait time! current timeout=100
段错误 (核心已转储)

x18-1, 2024-07-05:
你知道怎样设置width、height、fps这些参数吗

biosbeta, 2024-07-05:
> 你知道怎样设置width、height、fps这些参数吗

profile_list.get_video_stream_profile 方法设置