orbbec/pyorbbecsdk #56: pipeline.get_camera_param() returns all values equal to zero

GioZannini, 2024-07-25:
I registered a bag file using UI OrbbecViewer, after that I wanted to read this file and extract point clouds from it leveraging the pyorbbec SDK, here the code:

```python
import open3d as o3d
from pyorbbecsdk import *

bag_path = "file.bag"
pipeline = Pipeline(bag_path)
playback = pipeline.get_playback()
device_info = playback.get_device_info()
print("Device info: ", device_info)
camera_param = pipeline.get_camera_param()
print("camera param: ", camera_param)

pipeline.start()

while True:
    frames = pipeline.wait_for_frames(100)
    if frames is None:
        print("no frame")
        continue
    else:
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        if depth_frame:
            print("depth ok")
        else:
            print("depth no")

        if color_frame:
            print("color ok")
        else:
            print("color no")
        points = frames.get_color_point_cloud(camera_param)
        xyz = np.array(points[:, :3])
        colors = np.array(points[:, 3:], dtype=np.uint8)
        pcd = convert_to_o3d_point_cloud(xyz, colors)

        view_point = {
                    "front" : [ -0.29296638885843013, -0.18191107447471108, -0.93865811453516312 ],
                    "lookat" : [ -90.498397743538007, -233.12308084594011, 1498.7031155443628 ],
                    "up" : [ 0.26577419263896368, -0.95853727145541245, 0.10281234243961222 ],
                    "zoom" : 0.57999999999999985
                }
        o3d.visualization.draw_geometries([pcd], **view_point)
pipeline.stop()
```
When I try to retrieve camera parameters from Pipeline object I obtain all values set to zero and for this reason I can't obtain the point cloud, here the output:
```bash
[07/25 09:02:28.200395][info][57963][Pipeline.cpp:51] Playback Pipeline init ...
Device info:  DeviceInfo(name=Femto Mega
, pid=1641
, vid=11205, uid=
, serial_number=CL2A141003P
, firmware_version=1.2.7
, connection_type=
, hardware_version=
, supported_min_sdk_version=
, device_type=0)
camera param:  <OBCameraParam depth_intrinsic < fx=1fy = 1 cx =320 cy=180 width=640 height=360 > 
 depth_distortion < k1=0 k2=0 k3=0 k4=0 k5=0 k6=0 p1=0 p2=0 > 
 rgb_intrinsic < fx=1fy = 1 cx =960 cy=540 width=1920 height=1080 > 
 rgb_distortion < k1=0 k2=0 k3=0 k4=0 k5=0 k6=0 p1=0 p2=0 > 
 transform < rot=[0, 0, 0, 0, 0, 0, 0, 0, 0]
 transform=[0, 0, 0]
[07/25 09:02:29.350405][info][57963][Pipeline.cpp:75] Pipeline destroyed! @0x55FDB64D83A0
[07/25 09:02:29.352117][info][57963][Pipeline.cpp:277] Pipeline start done!
depth ok
color ok
[07/25 09:02:29.750769][warning][59355][Pipeline.cpp:341] Pipeline source frameset queue fulled, drop the oldest frame!
```
I managed to visualize the point cloud hard coding the parameters of cx, cy, fx, fy, width and height of both depth_intrinsic and rgb_intrinsic, here is the code:
```python
camera_param.depth_intrinsic.height = depth_frame.get_height()
camera_param.depth_intrinsic.width = depth_frame.get_width()
camera_param.depth_intrinsic.cx = depth_frame.get_width() // 2
camera_param.depth_intrinsic.cy = depth_frame.get_height() // 2
camera_param.depth_intrinsic.fx = 1
camera_param.depth_intrinsic.fy = 1
camera_param.rgb_intrinsic.height = color_frame.get_height()
camera_param.rgb_intrinsic.width = color_frame.get_width()
camera_param.rgb_intrinsic.cx = color_frame.get_width() // 2
camera_param.rgb_intrinsic.cy = color_frame.get_height() // 2
camera_param.rgb_intrinsic.fx = 1
camera_param.rgb_intrinsic.fy = 1
```

How can I obtain the parameters from the camera? Why when I ran the first snippet of code I retrieve all values equal to zero?

TIMESTICKING, 2025-02-15:
Hi, I'm also encountering this problem, did you take care of it?

TIMESTICKING, 2025-02-15:
seems like in their example "save_pointcloud_by_o3d.py", the function works well

Navifra-Brad, 2025-03-16:
The code doesn't work properly because it wasn't designed to run with a .bag file.

To replay a .bag file recorded using the viewer,

```
playback = pipeline.get_playback()
```
is required, but even after modifying the save_pointcloud_by_o3d.py code to include this, the camera parameters still aren't being read properly...

Has anyone solved this issue?

YipKo, 2025-04-01:
The issue is that the Femto series does not support hardware D2C and has been switched to software D2C. 
Try using the code as follows:
```python
config = Config()
config.set_align_mode(OBAlignMode.SW_MODE) # using software D2C
# pipeline.enable_frame_sync() # optional but suggest
pipeline.start(config)
```