orbbec/pyorbbecsdk #57: ModuleNotFoundError: No module named 'pyorbbecsdk'`

JakkapongW, 2024-07-30:
I followed the steps in the **Orbbec SDK Python Wrapper Documentation**, but I am still getting the error **`ModuleNotFoundError: No module named 'pyorbbecsdk'`** after trying to run a file from the Examples.

Note: I am using VS Code with Python 3.10.4.

tushardmaske, 2024-07-30:
cd into pyorbbecsdk and paste commands below in the terminal 

export PYTHONPATH=$PYTHONPATH:$(pwd)/install/lib/
sudo bash ./scripts/install_udev_rules.sh
sudo udevadm control --reload-rules && sudo udevadm trigger

OR 

you can put these commands in .sh file (for ex setup.sh) and every-time just do "source setup.sh"

JakkapongW, 2024-07-30:
@tushardmaske  Thank you for you reply
Is this code for Linux? Do you have one for Windows?

5d6d0011, 2024-08-04:
Me too. On Windows, I have tried various methods, but still got ModuleNotFoundError: No module named 'pyorbbecsdk'

JunJie-zhang-o, 2024-08-08:
import sys
sys.path.append("your pyorbbecsdk.so path")

you can try this on the header of the py file

Withey958, 2024-09-04:
I got this working eventually but documentation could probably do with updating.

zhonghong322, 2024-09-09:
@JakkapongW @5d6d0011 which sample？ if you use save_pointcloud_to_disk.py or save_pointcloud_to_disk_by_filter.py , need install plyfile.   

When installing plyfile with pip install plyfile, do not use a VPN, as it may cause installation issues.

wjstarrsiii, 2024-10-03:
Hello.  I am also unable to get this working with Windows.  I have followed all instructions, compiled the DLLs and pyd files and placed them in examples folder, but I still get ModuleNotFound.

I have done the same on Archlinux and after updating PYTHONPATH to include the sdk install\lib directory I do not have this issue.

Is there any update on why this is not working on Windows?

Thanks

wjstarrsiii, 2024-10-03:
I have solved this in my Windows installation.  I found that the pyd file was being compiled for a different version of python than what I had in my virtual environment.  I deleted the entire pyorbbecsdk directory and cloned it fresh.  I then followed the instructions.  I used CMake-GUI, but I launched it from the command line within the venv after installing the requirements.  "C:\ProgramFiles\CMake\bin\cmake-gui.exe".  After configuring, generating, then building and installing with Visual Studio and copying the files over, the pyd correctly was built to cp311 and the examples all are running.

ChickenChiang, 2024-12-13:
@wjstarrsiii this worked for me, thanks!

kerby2000, 2025-01-21:
I'm getting the same error
"I found that the pyd file was being compiled for a different version of python"
@wjstarrsiii 
How did you find this?

wjstarrsiii, 2025-01-21:
By looking at the python version in the built drivers (pyorbbecsdk-1.3.1-py3.8...) when my virtual environment for the project was in python 3.10.  I realized that I had built the driver with my default system python version of 3.8 instead of doing it inside my virtual environment.

Abrahamh08, 2025-03-08:
What "solved" this for me was doing the Windows equivalent of setting the PYTHON_PATH like in their env.sh. This is not a viable solution since my python manager uv re-syncs every run. There has to be a better way of setting up the package.

Abrahamh08, 2025-03-10:
I went ahead and re-did the package setup, and as a bonus it pulls the files for you when you pip install so you don't have to do anything manually: https://github.com/Abrahamh08/pyorbbecsdk/tree/trunk-v2-main

I'm keeping this trunk on the latest working v2 releases. Latest working is v2.0.7. When v2.0.9 is fixed or there is a new release, I will submit a pull request to v2-main as my setup makes more sense and doesn't rely on PYTHONPATH

jaytjoa, 2025-08-28:
I can confirm that the issue with uv is still there. Have you solved this one @Abrahamh08 ?