orbbec/pyorbbecsdk #60: gemini 335L 硬件对齐

JiXAllen, 2024-08-22:
gemini 335L 设置硬件对齐无frame输出 `config.set_align_mode(OBAlignMode.HW_MODE)`

Hyunmin-H, 2024-08-23:
The same thing is happening to me too.
I used gemini 335, 336.

zhonghong322, 2024-08-23:
@jian-dong 看下。

xcy2011sky, 2024-09-04:
@ZhangHangTao 感谢你使用我们的产品，在使用过程中不便，欢迎提供具体的反馈意见，我们会安排人和您沟通。您后续有产品的建议，可以发送邮件xuchongyan@orbbec.com; 他会及时反馈您的建议！！
但是社区是公开网络场所，希望使用合理的表述；

zhonghong322, 2024-09-07:
@JiXAllen @Hyunmin-H  Sorry for the late reply.
1、First, determine your firmware version. Low-version firmware does not support hardware D2C. You can use OrbbecViewer to check the firmware version. As of today, the latest firmware for the G330 device is 1.3.25. You can use OrbbecViewer OTA to upgrade the firmware to this version.
2、After upgrading the firmware, you can use OrbbecViewer to check the Depth and Color resolutions that support hardware D2C. As shown in the following figure: 
[image]

Then set the supported resolutions in the OrbbecSDKConfig_v1.0.xml file. For example:
[image]

3、Finally, run the sample depth_color_sync_align_viewer.py ，as in windows ，Run the following command under the sample path.
python depth_color_sync_align_viewer.py HW

JiXAllen, 2024-09-08:
我的固件版本是1.3.25，但是没有OrbbecViewer没有D2C(HW)选项

zhonghong322, 2024-09-09:
> 我的固件版本是1.3.25，但是没有OrbbecViewer没有D2C(HW)选项

更新OrbbecViewer v1.10.12版本
https://github.com/orbbec/OrbbecSDK/releases/tag/v1.10.12

JiXAllen, 2024-09-09:
使用OrbbecViewer v1.10.12版本打开D2C(HW)后，无数据输出，和pysdk情况一样。

zhonghong322, 2024-09-09:
> 使用OrbbecViewer v1.10.12版本打开D2C(HW)后，无数据输出，和pysdk情况一样。（不知道为什么图片加不上comment）
> 
> 此外，请问如何关闭启动pipeline后输出的大段info log

我这里测试是正常的，你的设备，最开始时的固件版本是什么, 是不是设备没有硬件D2C的参数？

[image]

JiXAllen, 2024-09-09:
折腾这个硬件对齐，我在OrbbecViewer中点了下重启设备，然后就再也连不上了。

2024-09-09 19:49:57.146 INFO  [6733] [loggerInit@18] **********************************************
2024-09-09 19:49:57.146 INFO  [6733] [loggerInit@19]  OrbbecViewer launched! Welcome!! 
2024-09-09 19:49:57.146 INFO  [6733] [loggerInit@20]  	- Version: V1.10.12
2024-09-09 19:49:57.146 INFO  [6733] [loggerInit@21]  	- Author: 
2024-09-09 19:49:57.146 INFO  [6733] [loggerInit@22]  	- E-Mail: 
2024-09-09 19:49:57.146 INFO  [6733] [loggerInit@23]  	- Company: orbbec
2024-09-09 19:49:57.146 INFO  [6733] [loggerInit@24]  	- Website: http://www.orbbec.com.cn/
2024-09-09 19:49:57.146 INFO  [6733] [loggerInit@25] **********************************************
[09/09 19:49:57.555311][info][6733][Context.cpp:68] Context created with config: /home/jiangxin/下载/OrbbecViewer_v1.10.12_202408090905_linux_x64_release/OrbbecSDKConfig_v1.0.xml
[09/09 19:49:57.555333][info][6733][Context.cpp:73] Work directory=/home/jiangxin/下载/OrbbecViewer_v1.10.12_202408090905_linux_x64_release, SDK version=v1.10.12-20240809-69b687f
[09/09 19:49:57.555359][info][6733][LinuxPal.cpp:32] createObPal: create LinuxPal!
[09/09 19:49:57.557278][info][6733][LinuxPal.cpp:143] Create PollingDeviceWatcher!
[09/09 19:49:57.557330][info][6733][DeviceManager.cpp:15] Current found device(s): (1)
[09/09 19:49:57.557344][info][6733][DeviceManager.cpp:24] 	- Name: Orbbec Gemini 335L, PID: 0x0804, SN/ID: , Connection: USB3.2
[09/09 19:49:57.557382][info][6733][DeviceManager.cpp:304] Enable net device enumeration: true
[09/09 19:49:58.559984][info][6733][DeviceManager.cpp:15] Current device(s) list: (1)
[09/09 19:49:58.560020][info][6733][DeviceManager.cpp:24] 	- Name: Orbbec Gemini 335L, PID: 0x0804, SN/ID: , Connection: USB3.2
[09/09 19:49:58.560660][error][6733][DeviceLibusb.cpp:112] failed to open usb device!  error: OB_USB_STATUS_ACCESS
[09/09 19:49:58.560831][warning][6733][EnumeratorLibusb.cpp:343] Execute failure! A std::exception has occurred!
	 - where:343#createUsbDevice
	 - msg:failed to open usb device!  error: OB_USB_STATUS_ACCESS
	 - type:St13runtime_error
[09/09 19:49:58.560861][warning][6733][EnumeratorLibusb.cpp:343] failed to create usb device at index: 8, url:4-2-7
[09/09 19:49:59.061079][error][6733][DeviceLibusb.cpp:112] failed to open usb device!  error: OB_USB_STATUS_ACCESS
[09/09 19:49:59.061185][warning][6733][EnumeratorLibusb.cpp:343] Execute failure! A std::exception has occurred!
	 - where:343#createUsbDevice
	 - msg:failed to open usb device!  error: OB_USB_STATUS_ACCESS
	 - type:St13runtime_error
[09/09 19:49:59.061211][warning][6733][EnumeratorLibusb.cpp:343] failed to create usb device at index: 8, url:4-2-7
[09/09 19:49:59.061233][warning][6733][ObException.hpp:40] usbEnumerator createUsbDevice failed!
2024-09-09 19:49:59.061 ERROR [6733] [OBSensorServer::connectToDevice@312] function:getDevice
args:index
message:usbEnumerator createUsbDevice failed!
type:1

jian-dong, 2024-09-09:
在scipts 下面有个脚本，sudo 运行一下

`sudo bash ./install_udev_rules.sh `

JiXAllen, 2024-09-09:
> 在scipts 下面有个脚本，sudo 运行一下
> 
> `sudo bash ./install_udev_rules.sh `

好的谢谢你，吓死我了。 最开始的估计版本没注意，固件至今更新过两次。
使用pysdk，OrbbecSDKConfig_v1.0.xml在哪呢？ 
我只找到SDK_ROS下的OrbbecSDKConfig_v1.0.xml配置文件，或者OrbbecViewer中的配置文件。

jian-dong, 2024-09-09:
这里https://github.com/orbbec/pyorbbecsdk/tree/main/config

JiXAllen, 2024-09-09:
> 因为这个是lib，不是exe，所以安装的话没有调用配置文件，你要给他指定路径。 下面这样,把`your_path`改成你的路径
> 
> ```python
> # ******************************************************************************
> #  Copyright (c) 2023 Orbbec 3D Technology, Inc
> #  
> #  Licensed under the Apache License, Version 2.0 (the "License");
> #  you may not use this file except in compliance with the License.  
> #  You may obtain a copy of the License at
> #  
> #      http:# www.apache.org/licenses/LICENSE-2.0
> #  
> #  Unless required by applicable law or agreed to in writing, software
> #  distributed under the License is distributed on an "AS IS" BASIS,
> #  WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
> #  See the License for the specific language governing permissions and
> #  limitations under the License.
> # ******************************************************************************
> import threading
> import time
> from typing import Optional
> 
> from pyorbbecsdk import *
> 
> device: Optional[Device] = None
> pipeline: Optional[Pipeline] = None
> 
> device_lock = threading.Lock()
> 
> 
> def start_stream():
>     global device
>     global pipeline
>     config = Config()
>     if device is None:
>         print("No device connected")
>         return
>     pipeline = Pipeline(device)
>     print("try to enable stream")
>     try:
>         profile_list = pipeline.get_stream_profile_list(OBSensorType.COLOR_SENSOR)
>         try:
>             color_profile: VideoStreamProfile = profile_list.get_video_stream_profile(1280, 0, OBFormat.RGB, 30)
>         except OBError as e:
>             print(e)
>             color_profile = profile_list.get_default_video_stream_profile()
>         config.enable_stream(color_profile)
>     except Exception as e:
>         print(e)
>     try:
>         profile_list = pipeline.get_stream_profile_list(OBSensorType.DEPTH_SENSOR)
>         try:
>             depth_profile: VideoStreamProfile = profile_list.get_video_stream_profile(640, 0, OBFormat.Y16, 30)
>         except OBError as e:
>             print(e)
>             depth_profile = profile_list.get_default_video_stream_profile()
>         config.enable_stream(depth_profile)
>     except Exception as e:
>         print(e)
>     print("try to start stream")
>     pipeline.start(config)
> 
> 
> def stop_stream():
>     global pipeline
>     if pipeline is None:
>         print("Pipeline is not started")
>         return
>     pipeline.stop()
>     pipeline = None
> 
> 
> def on_device_connected_callback(device_list: DeviceList):
>     if device_list.get_count() == 0:
>         return
>     global device
>     print("Device connected")
>     with device_lock:
>         if device is not None:
>             print("Device is already connected")
>             return
>         print("Try to get device")
>         device = device_list.get_device_by_index(0)
>         start_stream()
> 
> 
> def on_device_disconnected_callback(device_list: DeviceList):
>     global device, pipeline
>     if device_list.get_count() == 0:
>         return
>     print("Device disconnected")
>     with device_lock:
>         device = None
>         pipeline = None
> 
> 
> def on_new_frame_callback(frame: Frame):
>     if frame is None:
>         return
>     print("{} frame, width={}, height={}, format={}, timestamp={}us".format(frame.get_type(),
>                                                                             frame.get_width(),
>                                                                             frame.get_height(),
>                                                                             frame.get_format(),
>                                                                             frame.get_timestamp_us()))
> 
> 
> def on_device_changed_callback(disconn_device_list: DeviceList, conn_device_list: DeviceList):
>     on_device_connected_callback(conn_device_list)
>     on_device_disconnected_callback(disconn_device_list)
> 
> 
> def main():
>     config_path = "/your_path/OrbbecSDKConfig_v1.0.xml"
>     ctx = Context(config_path)
>     ctx.set_device_changed_callback(on_device_changed_callback)
>     device_list = ctx.query_devices()
>     on_device_connected_callback(device_list)
>     global pipeline, device
> 
>     while True:
>         try:
>             with device_lock:
>                 if pipeline is not None and device is not None:
>                     frames: FrameSet = pipeline.wait_for_frames(100)
>                 else:
>                     continue
>             if frames is None:
>                 time.sleep(0.001)
>                 continue
>             color_frame = frames.get_color_frame()
>             depth_frame = frames.get_depth_frame()
>             on_new_frame_callback(color_frame)
>             on_new_frame_callback(depth_frame)
>         except KeyboardInterrupt:
>             break
>     if pipeline is not None:
>         pipeline.stop()
> 
> 
> if __name__ == "__main__":
>     main()
> ```

谢谢，我想问的正是这个接口呢，结果都没文档。还有这个作为example不比那些强嘛

JiXAllen, 2024-09-09:
我知道了，我设定的图像尺寸1280，HW同步不支持该高分辨率同步，设定640后有输出了。
但是，Orbberviewer中依旧没有输出，该问题未知。

zhonghong322, 2024-09-09:
> 我知道了，我设定的图像尺寸1280，HW同步不支持该高分辨率同步，设定640后有输出了。 但是，Orbberviewer中依旧没有输出，该问题未知。

你说OrbbecViewer 没有输出哪个分辨率，截个图。

JiXAllen, 2024-09-09:
我不知道为什么发不了图片，图片加载进来Uploading后会自己消失，要不您给个地址我发邮箱。

zhonghong322, 2024-09-09:
> 我不知道为什么发不了图片，图片加载进来Uploading后会自己消失，要不您给个地址我发邮箱。

zhonghong@orbbec.com

JiXAllen, 2024-09-10:
> JiXAllen

As the solution provided by @zhonghong322 , you can try to update the firmware version, and do not use the high resolution(such as 1280).

JiXAllen, 2024-09-10:
> The same thing is happening to me too. I used gemini 335, 336.

As the solution provided by @zhonghong322 , you can try to update the firmware version, and do not use the high resolution(such as 1280).

JiXAllen, 2024-09-10:
此外，请问pipeline开始前几帧白平衡不对(发黄)有什么解决办法?

jian-dong, 2024-09-10:
> 此外，请问pipeline开始前几帧白平衡不对(发黄)有什么解决办法?

我关闭了这个issue，有什么问题从新开个issue

closed 2024-09-10 by jian-dong