orbbec/pyorbbecsdk #64: 当我用ubuntu系统make的时候出错了

apan666555, 2024-09-09:
/usr/bin/ld: cannot find -lOrbbecSDK collect2: error: ld returned 1 exit status make[2]: *** [CMakeFiles/pyorbbecsdk.dir/build.make:265: pyorbbecsdk.cpython-38-x86_64-linux-gnu.so] Error 1 make[1]: *** [CMakeFiles/Makefile2:104: CMakeFiles/pyorbbecsdk.dir/all] Error 2 make: *** [Makefile:141: all] Error 2

jian-dong, 2024-09-09:
是不是在windows下载，linux上解压了。

apan666555, 2024-09-09:
是，我又重新下载了，好了

---- 回复的原邮件 ----
| 发件人 | Joe ***@***.***> |
| 日期 | 2024年09月09日 16:18 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [orbbec/pyorbbecsdk] 当我用ubuntu系统make的时候出错了 (Issue #64) |

是不是在windows下载，linux上解压了。

zhonghong322, 2024-09-09:
> 是，我又重新下载了，好了
> […](#)

windows 下解压，在拷贝到linux，符号链接丢了，需要再linux下解压。

closed 2024-09-09 by jian-dong