orbbec/pyorbbecsdk #65: 如何不在终端terminal显示日志信息

apan666555, 2024-09-09:
指令是什么？

jian-dong, 2024-09-09:
context.set_logger_level(OBLogLevel.NONE)

jian-dong, 2024-09-09:
参考这个 https://github.com/orbbec/pyorbbecsdk/blob/main/examples/pointcloud_filter_o3d.py

apan666555, 2024-09-10:
我的意思是，我每次运行相机都会在我终端显示信息，我不想显示这些信息，上面您那条指令不好使
[09/10 10:18:15.045875][info][143705][Context.cpp:68] Context created with config: /home/lin/jiangpan/pyorbbecsdk-main/config/OrbbecSDKConfig_v1.0.xml
[09/10 10:18:15.045889][info][143705][Context.cpp:73] Work directory=/home/lin/jiangpan/pyorbbecsdk-main/examples, SDK version=v1.10.12-20240809-69b687f
[09/10 10:18:15.045901][info][143705][LinuxPal.cpp:32] createObPal: create LinuxPal!
[09/10 10:18:15.047967][info][143705][LinuxPal.cpp:143] Create PollingDeviceWatcher!
[09/10 10:18:15.048025][info][143705][DeviceManager.cpp:15] Current found device(s): (1)
[09/10 10:18:15.048047][info][143705][DeviceManager.cpp:24] 	- Name: Femto Bolt, PID: 0x066b, SN/ID: CL8FC31001N, Connection: USB3.1
[09/10 10:18:15.048094][info][143705][Pipeline.cpp:15] Try to create pipeline with default device.
[09/10 10:18:15.048186][info][143705][FemtoBoltUvcDevice.cpp:23] FemtoBoltUvcDevice init ...
[09/10 10:18:15.048298][info][143705][FemtoBoltUvcDevice.cpp:122] Create command start!
[09/10 10:18:15.049341][info][143705][MSDEConverterDevice.cpp:726] Succeed to load depth engine plugin
[09/10 10:18:15.358000][info][143705][AbstractDevice.cpp:121] 	- Firmware version: 1.0.9
[09/10 10:18:15.358030][info][143705][FemtoBoltUvcDevice.cpp:281] Create command done!
[09/10 10:18:15.358060][info][143705][FemtoBoltUvcDevice.cpp:402] init sensor map start!
[09/10 10:18:15.358085][info][143705][FemtoBoltUvcDevice.cpp:429] init sensor map done!
[09/10 10:18:15.358368][info][143705][FemtoBoltUvcDevice.cpp:285] Init depth process param start!
[09/10 10:18:15.700932][info][143705][MSDEConverterDevice.cpp:777] got nvram data succeed.
[09/10 10:18:16.120231][info][143705][FemtoBoltUvcDevice.cpp:359] setNvramDataStreamStopFunc succeed
[09/10 10:18:16.120357][info][143705][FemtoBoltUvcDevice.cpp:398] Init depth process param done!
[09/10 10:18:16.121077][info][143705][FemtoBoltUvcDevice.cpp:39] FemtoBoltUvcDevice init done!
[09/10 10:18:16.121113][info][143705][DeviceManager.cpp:150] Device created successfully! Name: Femto Bolt, PID: 0x066b, SN/ID: CL8FC31001N
[09/10 10:18:16.121203][info][143705][Pipeline.cpp:47] Pipeline created with device: {name: Femto Bolt, sn: CL8FC31001N}, @0x443A420
[09/10 10:18:16.121988][info][143705][FemtoBoltUvcDevice.cpp:529] Depth sensor has been created!
[09/10 10:18:16.122123][info][143705][VideoSensor.cpp:384] Query stream profile! size= 14, SensorType=OB_SENSOR_DEPTH
[09/10 10:18:16.122148][info][143705][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 640, height: 576, fps: 15}
[09/10 10:18:16.122168][info][143705][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 1024, height: 1024, fps: 15}
[09/10 10:18:16.122187][info][143705][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 1024, height: 1024, fps: 5}
[09/10 10:18:16.122206][info][143705][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 640, height: 576, fps: 30}
[09/10 10:18:16.122224][info][143705][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 320, height: 288, fps: 30}
[09/10 10:18:16.122242][info][143705][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 640, height: 576, fps: 25}
[09/10 10:18:16.122261][info][143705][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 320, height: 288, fps: 25}
[09/10 10:18:16.122280][info][143705][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 320, height: 288, fps: 15}
[09/10 10:18:16.122297][info][143705][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 640, height: 576, fps: 5}
[09/10 10:18:16.122315][info][143705][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 320, height: 288, fps: 5}
[09/10 10:18:16.122333][info][143705][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 512, height: 512, fps: 30}
[09/10 10:18:16.122352][info][143705][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 512, height: 512, fps: 25}
[09/10 10:18:16.122383][info][143705][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 512, height: 512, fps: 15}
[09/10 10:18:16.122403][info][143705][VideoSensor.cpp:386]  - {type: OB_STREAM_DEPTH, format: OB_FORMAT_Y16, width: 512, height: 512, fps: 5}
[09/10 10:18:16.123990][info][143705][FemtoBoltUvcDevice.cpp:574] Color sensor has been created!
[09/10 10:18:16.125393][info][143705][VideoSensor.cpp:384] Query stream profile! size= 84, SensorType=OB_SENSOR_COLOR
[09/10 10:18:16.125419][info][143705][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1920, height: 1080, fps: 15}
[09/10 10:18:16.125438][info][143705][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 720, fps: 30}
[09/10 10:18:16.125454][info][143705][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 720, fps: 25}
[09/10 10:18:16.125469][info][143705][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 720, fps: 15}
[09/10 10:18:16.125487][info][143705][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 720, fps: 5}
[... cut ...]
[09/10 10:18:16.125558][info][143705][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 720, fps: 5}
[09/10 10:18:16.125576][info][143705][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 960, fps: 30}
[09/10 10:18:16.125595][info][143705][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 960, fps: 25}
[09/10 10:18:16.125613][info][143705][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 960, fps: 15}