orbbec/pyorbbecsdk #67: Synchronisation Config

Withey958, 2024-09-11:
I am using two Astra 2 devices with a Multi-Camara Sync Hub Dev on windows 11. I am having issues with updating the settings. Have updated settings as shown below (it is recommended for astra 2 the `irTriggerSignalInDelay` is at least 4000). I assumed this corresponded to `depth_delay_us`. I may be wrong:

```
{
    "devices": [
        {
            "serial_number": "XXXXXXXX",
            "config": {
                "mode": "PRIMARY",
                "depth_delay_us": 10000,
                "color_delay_us": 0,
                "trigger_to_image_delay_us": 0,
                "trigger_out_enable": true,
                "trigger_out_delay_us": 0,
                "frames_per_trigger": 1
            }
        },
        {
            "serial_number": "XXXXXXX",
            "config": {
                "mode": "SECONDARY",
                "depth_delay_us": 10000,
                "color_delay_us": 0,
                "trigger_to_image_delay_us": 0,
                "trigger_out_enable": true,
                "trigger_out_delay_us": 0,
                "frames_per_trigger": 1
            }
        }
    ]
}
```

When I try to update this using example code: `multicamera_recorder.py` to check that the cameras have accepted these changes the I get a warning. `[info][22352][DeviceSyncConfigurator.cpp:34] New sync config is same as current device sync config, the upgrade process would not execute!`. But when I print what I am trying to change using:

```
        sync_config: OBMultiDeviceSyncConfig = device.get_multi_device_sync_config()

        sync_config_json = multi_device_sync_config[serial_number]
        sync_config = device.get_multi_device_sync_config()
        sync_config.mode = sync_mode_from_str(sync_config_json["config"]["mode"])
        sync_config.color_delay_us = sync_config_json["config"]["color_delay_us"]
        sync_config.depth_delay_us = sync_config_json["config"]["depth_delay_us"]
        sync_config.trigger_out_enable = sync_config_json["config"][
            "trigger_out_enable"
        ]
        sync_config.trigger_out_delay_us = sync_config_json["config"][
            "trigger_out_delay_us"
        ]
        sync_config.frames_per_trigger = sync_config_json["config"][
            "frames_per_trigger"
        ]
        device.set_multi_device_sync_config(sync_config)
        print(f"Device {serial_number} sync config: {sync_config}")
        print(
            f"""set sync config:{[sync_config.color_delay_us,
        sync_config.depth_delay_us,
        sync_config.frames_per_trigger,
        sync_config.mode,
        sync_config.trigger_out_delay_us,
        sync_config.trigger_out_enable,
        sync_config.trigger_to_image_delay_us]}"""
        )

        got_sync_config: OBMultiDeviceSyncConfig = device.get_multi_device_sync_config()

        print(
            f"""got sync config:{[got_sync_config.color_delay_us,
        got_sync_config.depth_delay_us,
        got_sync_config.frames_per_trigger,
        got_sync_config.mode,
        got_sync_config.trigger_out_delay_us,
        got_sync_config.trigger_out_enable,
        got_sync_config.trigger_to_image_delay_us]}"""
        )
``` 

I get for cam1:

```
[info][24852][DeviceSyncConfigurator.cpp:34] New sync config is same as current device sync config, the upgrade process would not execute!
set sync config:[0, 4000, 1, <OBMultiDeviceSyncMode.PRIMARY: 4>, 0, True, 1]
got sync config:[0, 1, 1, <OBMultiDeviceSyncMode.PRIMARY: 4>, 0, True, 1]
```
and for cam 2:
```
[info][24852][DeviceSyncConfigurator.cpp:34] New sync config is same as current device sync config, the upgrade process would not execute!
set sync config:[0, 4000, 1, <OBMultiDeviceSyncMode.SECONDARY_SYNCED: 16>, 0, True, 1]
got sync config:[0, 1, 1, <OBMultiDeviceSyncMode.SECONDARY_SYNCED: 16>, 0, True, 1]
```

Am I trying to change the correct setting?  And why wont this update (are there limits I am not seeing)? If I am not doing something incorrectly there may be an bug within the examples.

zhonghong322, 2024-09-14:
The purpose of setting a delay for Astra 2 is to prevent laser interference between devices, and the delay should be at least 4000us. If your Multi-Camara Sync Hub Dev is star type , the parameters to configure are as follows:
{
    "devices": [
        {
            "serial_number": "XXXXXXXX",
            "config": {
                "mode": "PRIMARY",
                "depth_delay_us": 0,
                "color_delay_us": 0,
                "trigger_to_image_delay_us": 0,
                "trigger_out_enable": true,
                "trigger_out_delay_us": -1,
                "frames_per_trigger": 1
            }
        },
        {
            "serial_number": "XXXXXXX",
            "config": {
                "mode": "SECONDARY",
                "depth_delay_us": 0,
                "color_delay_us": 0,
                "trigger_to_image_delay_us": 4000,
                "trigger_out_enable": true,
                "trigger_out_delay_us": 0,
                "frames_per_trigger": 1
            }
        }
    ]
}

If using a daisy-chain-type sync hub, the parameters to configure are as follows:
{
    "devices": [
        {
            "serial_number": "XXXXXXXX",
            "config": {
                "mode": "PRIMARY",
                "depth_delay_us": 0,
                "color_delay_us": 0,
                "trigger_to_image_delay_us": 0,
                "trigger_out_enable": true,
                "trigger_out_delay_us": -1,
                "frames_per_trigger": 1
            }
        },
        {
            "serial_number": "XXXXXXX",
            "config": {
                "mode": "SECONDARY",
                "depth_delay_us": 0,
                "color_delay_us": 0,
                "trigger_to_image_delay_us": 0,
                "trigger_out_enable": true,
                "trigger_out_delay_us": -1,
                "frames_per_trigger": 1
            }
        }
    ]
}

zhonghong322, 2024-09-14:
Trigger Out Delay: -1 means the output time is based on the camera's default exposure time.

zhonghong322, 2024-09-14:
[image]

Withey958, 2024-09-16:
Thanks. Thats exactly what I needed. To clarify I am using the daisy chain.