orbbec/pyorbbecsdk #74: align_filter_viewer.py example does not work

lionbur, 2024-10-09:
It fails on line 89:
```python
            frames = align_filter.process(frames)
```

Calling `process` always return `None`.

Running on Raspberry Pi 5 with latest Raspberry Pi OS 64-bit with Orbbec Astra 2 attached via USB 3 port.

Is there a workaround for this (aligning color and depth frames with some code)?

Full output:
```
[10/09 17:30:36.112454][info][30149][Context.cpp:68] Context created with config: /home/admin/work/pyorbbecsdk/config/OrbbecSDKConfig_v1.0.xml
[10/09 17:30:36.112491][info][30149][Context.cpp:73] Work directory=/home/admin/work/pyorbbecsdk/examples, SDK version=v1.10.12-20240809-69b687f
[10/09 17:30:36.112500][info][30149][LinuxPal.cpp:32] createObPal: create LinuxPal!
[10/09 17:30:36.243129][info][30149][LinuxPal.cpp:143] Create PollingDeviceWatcher!
[10/09 17:30:36.243157][info][30149][DeviceManager.cpp:15] Current found device(s): (1)
[10/09 17:30:36.243163][info][30149][DeviceManager.cpp:24] 	- Name: Orbbec Astra 2, PID: 0x0660, SN/ID: AARJ73P0065, Connection: USB2.0
[10/09 17:30:36.243169][info][30149][Pipeline.cpp:15] Try to create pipeline with default device.
[10/09 17:30:36.355937][info][30149][Astra2Device.cpp:343] Create command done!
[10/09 17:30:36.357892][info][30149][Astra2Device.cpp:410] init sensor map start!
[10/09 17:30:36.359873][info][30149][AbstractDevice.cpp:121] 	- Firmware version: 2.8.20
[10/09 17:30:36.394496][info][30149][Astra2Device.cpp:157] Astra2 device created! PID: 0x0660, SN: AARJ73P0065
[10/09 17:30:36.394510][info][30149][DeviceManager.cpp:150] Device created successfully! Name: Orbbec Astra 2, PID: 0x0660, SN/ID: AARJ73P0065
[10/09 17:30:36.394531][info][30149][Pipeline.cpp:47] Pipeline created with device: {name: Orbbec Astra 2, sn: AARJ73P0065}, @0x2EF1E710
[10/09 17:30:36.403385][info][30149][Astra2Device.cpp:625] Rgb sensor has been created!
[10/09 17:30:36.410276][info][30149][VideoSensor.cpp:384] Query stream profile! size= 120, SensorType=OB_SENSOR_COLOR
[10/09 17:30:36.410291][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 720, fps: 30}
[10/09 17:30:36.410296][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 320, height: 180, fps: 30}
[10/09 17:30:36.410300][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 320, height: 180, fps: 15}
[10/09 17:30:36.410305][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 320, height: 180, fps: 10}
[10/09 17:30:36.410309][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 320, height: 180, fps: 5}
[10/09 17:30:36.410312][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 320, height: 240, fps: 30}
[10/09 17:30:36.410316][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 320, height: 240, fps: 15}
[10/09 17:30:36.410320][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 320, height: 240, fps: 10}
[10/09 17:30:36.410324][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 320, height: 240, fps: 5}
[10/09 17:30:36.410328][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 400, height: 300, fps: 30}
[10/09 17:30:36.410332][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 400, height: 300, fps: 15}
[10/09 17:30:36.410336][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 400, height: 300, fps: 10}
[10/09 17:30:36.410340][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 400, height: 300, fps: 5}
[10/09 17:30:36.410344][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 640, height: 360, fps: 30}
[10/09 17:30:36.410349][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 640, height: 360, fps: 15}
[10/09 17:30:36.410357][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 640, height: 360, fps: 10}
[10/09 17:30:36.410361][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 640, height: 360, fps: 5}
[10/09 17:30:36.410366][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 640, height: 480, fps: 30}
[10/09 17:30:36.410371][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 640, height: 480, fps: 15}
[10/09 17:30:36.410374][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 640, height: 480, fps: 10}
[10/09 17:30:36.410378][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 640, height: 480, fps: 5}
[10/09 17:30:36.410382][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 800, height: 600, fps: 10}
[10/09 17:30:36.410386][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 800, height: 600, fps: 5}
[10/09 17:30:36.410390][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 720, fps: 10}
[10/09 17:30:36.410394][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_YUYV, width: 1280, height: 720, fps: 5}
[10/09 17:30:36.410397][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 320, height: 240, fps: 30}
[10/09 17:30:36.410401][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 320, height: 240, fps: 15}
[10/09 17:30:36.410405][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 320, height: 240, fps: 10}
[10/09 17:30:36.410408][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 320, height: 240, fps: 5}
[10/09 17:30:36.410412][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 640, height: 360, fps: 30}
[10/09 17:30:36.410416][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 640, height: 360, fps: 15}
[10/09 17:30:36.410419][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 640, height: 360, fps: 10}
[10/09 17:30:36.410423][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 640, height: 360, fps: 5}
[10/09 17:30:36.410427][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 640, height: 480, fps: 30}
[10/09 17:30:36.410431][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 640, height: 480, fps: 15}
[10/09 17:30:36.410434][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 640, height: 480, fps: 10}
[10/09 17:30:36.410438][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 640, height: 480, fps: 5}
[10/09 17:30:36.410442][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 800, height: 600, fps: 30}
[10/09 17:30:36.410445][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 800, height: 600, fps: 15}
[10/09 17:30:36.410449][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 800, height: 600, fps: 10}
[10/09 17:30:36.410453][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 800, height: 600, fps: 5}
[10/09 17:30:36.410456][info][30149][VideoSensor.cpp:386]  - {type: OB_STREAM_COLOR, format: OB_FORMAT_MJPG, width: 1280, height: 720, fps: 15}
[... 129 lines cut ...]
[10/09 17:30:36.603716][warning][30149][FrameProcessingBlock.cpp:43] Execute failure! A libobsensor_exception has occurred!
	 - where:43#process
	 - msg:Can not find matched camera param!
	 - type:N11libobsensor31unsupported_operation_exceptionE
[10/09 17:30:36.603722][error][30149][FrameProcessingBlock.cpp:43] Process failed!
[10/09 17:30:36.616958][info][30149][Pipeline.cpp:411] Try to stop pipeline!
[10/09 17:30:36.617005][info][30149][Pipeline.cpp:383] Try to stop streams!
[... cut ...]
[10/09 17:30:36.644496][info][30149][ObLibuvcDevicePort.cpp:71] ~ObLibuvcDevicePort done
[10/09 17:30:36.646166][info][30149][Context.cpp:84] Context destroyed
[10/09 17:30:36.646286][warning][30162][VideoSensor.cpp:443] This frame will be dropped because jpg format verification failure! @OB_SENSOR_COLOR [**3 logs in 148ms, last: 17:30:36.554363**]
[10/09 17:30:36.646398][warning][30161][VideoSensor.cpp:453] This frame will be dropped because the data size does not match the expectation! size=36736, expected=210000 @OB_SENSOR_DEPTH [**8 logs in 206ms, last: 17:30:36.508084**]
```

zhonghong322, 2024-10-14:
Right now, The align_filter_viewer.py sample only supports the Gemini 330 series. For Astra2, please use the depth_color_sync_align_viewer.py sample.

mightypanda, 2024-10-26:
Can we build and use the pyorbbecsdk for a Gemini 2 on a Raspberry Pi 5 without problems?