orbbec/pyorbbecsdk #76: Accessing depth to 3d point with transformationDepthToPointCloud()

mightypanda, 2024-10-10:
Hello,
Is it possible to access the [transformationDepthToPointCloud()](https://github.com/orbbec/OrbbecSDK/blob/10823024d7947a24846acf32ac4092e836f2d727/include/libobsensor/hpp/Utils.hpp#L125C17-L125C48) function from the SDK ?

I have a similar need as this thread, but doing it in python.
https://github.com/orbbec/OrbbecSDK/issues/39#issuecomment-1896421951

Thank you

nbubis, 2024-12-25:
@mightypanda Did you end up solving this issue?

mightypanda, 2025-02-18:
> [@mightypanda](https://github.com/mightypanda) Did you end up solving this issue?

Hello @nbubis , i ended up going a totally different way, using the pointcloud directly and "splatting" it into a 2d image for blob detection. Optimization took some time but it seems to work rather well.

importnumpy, 2025-04-29:
@mightypanda Hello, Could you please share the implementation code?