orbbec/pyorbbecsdk #78: How to reset/reboot the camera

elevenjiang1, 2024-10-12:
We encountered a power supply issue when using the Bolt device. The details are as follows:

Power Setup:

The Bolt is connected to a computer via USB. If the USB power supply is insufficient, the Bolt flashes a yellow light.
The robot provides a 12V power supply.
Issue Description:

Step 1: The computer is powered on, and the Bolt is powered via USB but not in use. At this point, the robot is not powered on, the Bolt is not receiving 12V power, and the status light is normal (white).
Step 2: The robot is powered on, and the 12V power supply to the Bolt is activated. However, the Bolt remains unused, and the status light stays normal (white).
Step 3: When we start using the Bolt, it flashes a yellow light. We suspect this occurs because the Bolt fails to switch correctly to 12V power and continues to rely on the USB power supply, which is insufficient.
Step 4: We disconnect the USB port via software or physically unplug the USB cable. The Bolt continues to receive 12V power.
Step 5: After reconnecting the USB or re-enabling it, the Bolt still defaults to USB power and continues flashing a yellow light, indicating insufficient power.
We suspect that since the 12V power supply was not interrupted, the Bolt keeps defaulting to USB power. We prefer not to reboot the robot to resolve the issue, as it would cause additional complications. Therefore, we would like to ask if there is a software-based method to reset the Bolt to avoid the need for a robot power cycle.

**********

我们在使用Bolt时遇到了供电问题，具体如下：

供电设置：

Bolt通过USB连接电脑，若USB供电不足，Bolt会闪黄灯。
机器人提供12V供电。
问题描述：

步骤1：电脑通电，Bolt通过USB供电，但并未被使用。此时，机器人未通电，Bolt未从12V供电，状态正常（白灯）。
步骤2：机器人通电，12V供电开启，但Bolt依然未被使用，状态正常（白灯）。
步骤3：开始使用Bolt时，Bolt闪黄灯。我们猜测是因为Bolt没有正确切换到12V供电，依然默认使用USB供电，导致供电不足。
步骤4：软件断开USB端口或拔掉USB，Bolt依然由12V供电。
步骤5：重新连接USB，Bolt仍然默认使用USB供电，继续显示供电不足的黄灯状态。
我们猜测由于12V供电未断开，Bolt一直默认从USB供电。我们不希望通过重启机器人来解决问题，因为这会引发其他麻烦。因此，想请教是否有软件方法可以重置Bolt，从而避免机器人重新上电。

zhonghong322, 2024-10-14:
I didn't encounter the power supply issue you mentioned, and the restart command is as follows:
device->reboot();

zhonghong322, 2024-10-14:
I couldn't reproduce the power supply issue you mentioned. The interface for restarting the camera in Python is as follows:
    pipeline = Pipeline()
    device = pipeline.get_device()
    device.reboot()