orbbec/pyorbbecsdk #86: the video stream became frozen

Sukeysun, 2024-11-07:
After running the program for an extended period, the video stream became frozen and unresponsive.
got log as below:

[11/07 16:00:23.202062][warning][1233299][Pipeline.cpp:326] Wait for frame timeout, you can try to increase the wait time! current timeout=100 [**35 logs in 6050ms**]
[11/07 16:00:25.215383][warning][1233405][Pipeline.cpp:290] Pipeline source frameset queue fulled, drop the oldest frame!
[11/07 16:00:28.217853][warning][1233405][Pipeline.cpp:290] Pipeline source frameset queue fulled, drop the oldest frame! [**18 logs in 3002ms**]
[11/07 16:00:34.384987][warning][1233450][Pipeline.cpp:290] Pipeline source frameset queue fulled, drop the oldest frame! [**35 logs in 6167ms, last: 16:00:34.057499**]
[11/07 16:00:35.323902][warning][1233444][Pipeline.cpp:326] Wait for frame timeout, you can try to increase the wait time! current timeout=100 [**1 logs in 12121ms, last: 16:00:23.323599**]
[11/07 16:00:46.385420][warning][1233499][Pipeline.cpp:290] Pipeline source frameset queue fulled, drop the oldest frame! [**73 logs in 12000ms, last: 16:00:46.235603**]

nivesh48, 2024-11-08:
@Sukeysun may be you can try modifying memory buffer and queue size in `OrbbecSDKConfig_v1.0.xml`?

Sukeysun, 2024-11-11:
Thank you for your reply
I have made adjustments to the queue size in the OrbbecSDKConfig_v1.0.xml, but it appears that the changes have not taken effect. Below is the code I am working with:
`   from pyorbbecsdk import *
    config_path = "./OrbbecSDKConfig_v1.0.xml"
    context = Context(config_path)
    config = Config()
    config.set_align_mode(OBAlignMode.SW_MODE)
    video_writer=None
    context.set_logger_level(OBLogLevel.ERROR)

    profile_list = pipeline.get_stream_profile_list(OBSensorType.COLOR_SENSOR)
    color_profile: VideoStreamProfile = profile_list.get_video_stream_profile(640, 0, OBFormat.MJPG, 6)
    config.enable_stream(color_profile)
    profile_depth_list = pipeline.get_stream_profile_list(OBSensorType.DEPTH_SENSOR)
    depth_profile: VideoStreamProfile = profile_depth_list.get_video_stream_profile(640, 0, OBFormat.Y16, 6)
    config.enable_stream(depth_profile)
    pipeline.enable_frame_sync()
    pipeline.start(config)
`

Sukeysun, 2024-11-11:
@nivesh48 
Thank you for your reply
I have made adjustments to the queue size in the OrbbecSDKConfig_v1.0.xml, but it appears that the changes have not taken effect. Below is the code I am working with:
`   from pyorbbecsdk import *
    config_path = "./OrbbecSDKConfig_v1.0.xml"
    context = Context(config_path)
    config = Config()
    config.set_align_mode(OBAlignMode.SW_MODE)
    video_writer=None
    context.set_logger_level(OBLogLevel.ERROR)

    profile_list = pipeline.get_stream_profile_list(OBSensorType.COLOR_SENSOR)
    color_profile: VideoStreamProfile = profile_list.get_video_stream_profile(640, 0, OBFormat.MJPG, 6)
    config.enable_stream(color_profile)
    profile_depth_list = pipeline.get_stream_profile_list(OBSensorType.DEPTH_SENSOR)
    depth_profile: VideoStreamProfile = profile_depth_list.get_video_stream_profile(640, 0, OBFormat.Y16, 6)
    config.enable_stream(depth_profile)
    pipeline.enable_frame_sync()
    pipeline.start(config)
`