orbbec/pyorbbecsdk #89: Depth suddenly change

Sukeysun, 2024-11-11:
My camera is 355L.  I noticed that the depth would suddenly change. In the first few frames, an object was at a distance of 1 to 2 meters from the camera, but in the next frame, it would abruptly jump to 0 or 4 meters. I tried post_filter according to [post_process.py](https://github.com/orbbec/pyorbbecsdk/blob/main/examples/post_process.py)。But it doesn't work。

sevenandseven, 2024-11-28:
> 我的相机是355L。我注意到深度会突然改变。在前几帧中，物体距离相机有1到2米，但在下一帧中，它会突然跳到0或4米。我根据[post_process.py](https://github.com/orbbec/pyorbbecsdk/blob/main/examples/post_process.py)尝试了 post_filter 。但是不起作用。

你好，我遇到了同样的问题，请问应该怎么解决？

sevenandseven, 2024-11-28:
> > 我的相机是355L。我注意到深度会突然改变。在前几帧中，物体距离相机有1到2米，但在下一帧中，它会突然跳到0或4米。我根据[post_process.py](https://github.com/orbbec/pyorbbecsdk/blob/main/examples/post_process.py)尝试了 post_filter 。但是不起作用。
> 
> 你好，我遇到了同样的问题，请问应该怎么解决？

您是什么场景下遇到这种情况的？这种情况是偶发现象，但是目前不清楚什么场景下会发生。

zhonghong322, 2024-12-02:
Can you provide the original image of the depth suddenly change，For example, continuously store 10 frames of depth images.
Save images by referring to this sample （save_image_to_disk.py）

Sukeysun, 2024-12-04:
@sevenandseven 当目标快速移动的时候，会出现这样的情况

> > > 我的相机是355L。我注意到深度会突然改变。在前几帧中，物体距离相机有1到2米，但在下一帧中，它会突然跳到0或4米。我根据[post_process.py](https://github.com/orbbec/pyorbbecsdk/blob/main/examples/post_process.py)尝试了 post_filter 。但是不起作用。
> > 
> > 
> > 你好，我遇到了同样的问题，请问应该怎么解决？
> 
> 您是什么场景下遇到这种情况的？这种情况是偶发现象，但是目前不清楚什么场景下会发生。

dbestPacksol, 2025-01-15:
@zhonghong322 

I am having the same issue with both the 2.06 and the latest (1/15/25) v2-main branch code with the Femto Mega over network connection. Every few frames the depth value will increase drastically and then settle back despite nothing happening. The example here has the camera pointing at a wall and had a spike in depth values.

Please let me know anything else to help resolve this issue.

[depth_images.zip](https://github.com/user-attachments/files/18430764/depth_images.zip)