orbbec/pyorbbecsdk #98: Property not found to register callback, propertyId: 205

YeWenxuan64, 2024-11-24:
嘤，在运行example里面关于深度的py文件时发生错误
Property not found to register callback, propertyId: 205
只运行color.py未发生错误

SDK版本：pyorbbecsdk-2.0.5（release）

编译方式：
cd pyorbbecsdk-2.0.5
python3 -m venv ./venv
source venv/bin/activate
pip3 install -r requirements.txt
mkdir build
cd build
cmake -Dpybind11_DIR=`pybind11-config --cmakedir` ..
make -j4
make install

设备：336L（固件版本Gemini330_Release_1.3.70）
平台：
rk3588
Debian12
环境：
python3.11
包如requirement.txt

zhonghong322, 2024-11-25:
这个propertyid :205  在1.3.70固件 不支持，不影响使用。

YeWenxuan64, 2024-11-25:
> 这个propertyid :205 在1.3.70固件 不支持，不影响使用。

可是运行后它终止了，我该怎么让它（代码，example里面的，比如depth.py）运行起来呢

YeWenxuan64, 2024-11-25:
这是去掉try，expect后的报错
File "/home/yewen/Softwares/pyorbbecsdk-2.0.5/examples/depth.py", line 48, in main
    profile_list = pipeline.get_stream_profile_list(OBSensorType.DEPTH_SENSOR)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
pyorbbecsdk.OBError: Property not found to register callback, propertyId: 205

zhonghong322, 2024-11-30:
> 这是去掉try，expect后的报错 File "/home/yewen/Softwares/pyorbbecsdk-2.0.5/examples/depth.py", line 48, in main profile_list = pipeline.get_stream_profile_list(OBSensorType.DEPTH_SENSOR) 空格空格空格^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ pyorbbecsdk.OBError: Property not found to register callback, propertyId: 205
我看下。

zhonghong322, 2024-11-30:
> 这是去掉try，expect后的报错 File "/home/yewen/Softwares/pyorbbecsdk-2.0.5/examples/depth.py", line 48, in main profile_list = pipeline.get_stream_profile_list(OBSensorType.DEPTH_SENSOR) [...]

谢谢你反馈这个问题，我在linux和arm下测试，确实会报错，问题已经修复，提交到v2-main。

YeWenxuan64, 2024-12-01:
已测试tag2.0.6，问题已解决，工作正常，感想修复

closed 2024-12-01 by YeWenxuan64