orbbec/pyorbbecsdk #102: Python stub generation fails

nbubis, 2024-12-24:
Sourcing env.sh and running the pybind11-stubgen command one gets:

```
$ pybind11-stubgen pyorbbecsdk
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Config.enable_stream : Invalid expression 'ob::StreamProfile'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Context.create_net_device : Invalid expression 'ob::Device'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Context.query_devices : Invalid expression 'ob::DeviceList'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Device.get_bool_property_range : Can't find/import 'OBBoolPropertyRange'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Device.get_calibration_camera_param_list : Invalid expression 'ob::CameraParamList'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Device.get_sensor : Invalid expression 'ob::Sensor'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Device.get_sensor_list : Invalid expression 'ob::SensorList'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.DeviceList.get_device_by_index : Invalid expression 'ob::Device'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.DeviceList.get_device_by_serial_number : Invalid expression 'ob::Device'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.DeviceList.get_device_by_uid : Invalid expression 'ob::Device'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Filter.process : Invalid expression 'ob::Frame'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Filter.push_frame : Invalid expression 'ob::Frame'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Frame.as_accel_frame : Invalid expression 'ob::AccelFrame'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Frame.as_color_frame : Invalid expression 'ob::ColorFrame'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Frame.as_depth_frame : Invalid expression 'ob::DepthFrame'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Frame.as_frame_set : Invalid expression 'ob::FrameSet'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Frame.as_gyro_frame : Invalid expression 'ob::GyroFrame'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Frame.as_ir_frame : Invalid expression 'ob::IRFrame'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Frame.as_points_frame : Invalid expression 'ob::PointsFrame'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Frame.as_video_frame : Invalid expression 'ob::VideoFrame'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Frame.get_data_pointer : Can't find/import 'capsule'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Pipeline.get_config : Invalid expression 'ob::Config'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Pipeline.get_playback : Invalid expression 'ob::Playback'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Pipeline.get_stream_profile_list : Invalid expression 'ob::StreamProfileList'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Pipeline.start : Invalid expression 'ob::Config'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Pipeline.switch_config : Invalid expression 'ob::Config'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Playback.start : Invalid expression '<OBMediaType.???: 1023>'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.PointCloudFilter.calculate : Invalid expression 'ob::Frame'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Sensor.get_recommended_filters : Invalid expression 'ob::OBFilterList'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Sensor.get_stream_profile_list : Invalid expression 'ob::StreamProfileList'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Sensor.start : Invalid expression 'ob::StreamProfile'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.Sensor.switch_profile : Invalid expression 'ob::StreamProfile'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.StreamProfile.as_accel_stream_profile : Invalid expression 'ob::AccelStreamProfile'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.StreamProfile.as_gyro_stream_profile : Invalid expression 'ob::GyroStreamProfile'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.StreamProfile.as_video_stream_profile : Invalid expression 'ob::VideoStreamProfile'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.VideoFrame.as_color_frame : Invalid expression 'ob::ColorFrame'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.VideoFrame.as_depth_frame : Invalid expression 'ob::DepthFrame'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.VideoFrame.as_ir_frame : Invalid expression 'ob::IRFrame'
pybind11_stubgen - [  ERROR] In pyorbbecsdk.VideoFrame.as_points_frame : Invalid expression 'ob::PointsFrame'
pybind11_stubgen - [WARNING] Raw C++ types/values were found in signatures extracted from docstrings.
Please check the corresponding sections of pybind11 documentation to avoid common mistakes in binding code:
 - https://pybind11.readthedocs.io/en/latest/advanced/misc.html#avoiding-cpp-types-in-docstrings
 - https://pybind11.readthedocs.io/en/latest/advanced/functions.html#default-arguments-revisited
```

Output of `pip freeze`:
```
av==14.0.1
filelock==3.16.1
fsspec==2024.12.0
Jinja2==3.1.5
MarkupSafe==3.0.2
mpmath==1.3.0
networkx==3.4.2
numpy==2.2.1
opencv-python==4.10.0.84
pybind11==2.13.6
pybind11-stubgen==2.5.1
pybind11_global==2.13.6
scipy==1.14.1
sympy==1.13.1
torch==2.5.1
typing_extensions==4.12.2
```
Python version is 3.11.11

c99z, 2024-12-26:
I also failed,the same reason.

githubuser-0, 2025-07-31:
[image]