orbbec/pyorbbecsdk #103: (Regression) Camera Parameters All Zeros in v2 SDK

KySmith1, 2024-12-24:
PyOrbbec SDK Version: 2.0.6 (Latest as of time of writing)
OrbbecSDK Version: 2.1.1(Latest as of time of writing)
Camera: Femto Bolt
Camera FW Version: 1.1.2 (Latest as of time of writing)

Platform: Ubuntu 22.04 (Docker) running on Ubuntu 24.04 (Host)

When attempting to get the camera parameters (intrinsics), the return values for `pipeline.get_camera_param()` as seen in Orbbec's example here are all zeros:

[image]

This blocks our implementation of the v2 SDK in Python. Is there an undocumented new method for getting camera parameters?

More info:
The Orbbec SDK via Orbbec Viewer appears to get intrinsics without an issue, so I suspect this is something to do with this package:

[image]

TIMESTICKING, 2025-02-15:
same here, did you figure it out?

TIMESTICKING, 2025-02-15:
seems like in their example "save_pointcloud_by_o3d.py", the function works well

importnumpy, 2025-03-16:
This is the same issue as the one in the following link: https://github.com/orbbec/pyorbbecsdk/issues/56

Has anyone found a solution to this?

YipKo, 2025-04-01:
@KySmith1 @TIMESTICKING @importnumpy 
The issue is that the Femto series does not support hardware D2C and has been switched to software D2C. 
Try using the code as follows:
```python
config = Config()
config.set_align_mode(OBAlignMode.SW_MODE) # using software D2C
# pipeline.enable_frame_sync() # optional but suggest
pipeline.start(config)
```

zhonghong322, 2025-05-22:
When using the v2-main branch, the recommended way to obtain intrinsic and extrinsic parameters is from the stream profile. You can refer to the sample [coordinate_transform.py](https://github.com/orbbec/pyorbbecsdk/blob/v2-main/examples/coordinate_transform.py).

 ~~~
    color_frame = color_frame.as_video_frame()
    depth_frame = depth_frame.as_video_frame()

    color_profile = color_frame.get_stream_profile()
    depth_profile = depth_frame.get_stream_profile()
    print("video profile:", color_profile.as_video_stream_profile())
    color_intrinsics = color_profile.as_video_stream_profile().get_intrinsic()
    color_distortion = color_profile.as_video_stream_profile().get_distortion()
    depth_intrinsics = depth_profile.as_video_stream_profile().get_intrinsic()
    depth_distortion = depth_profile.as_video_stream_profile().get_distortion()

    extrinsic = depth_profile.get_extrinsic_to(color_profile)
~~~