orbbec/pyorbbecsdk #109: Support for 315Lg?

peter-mitrano-bg, 2025-02-03:
I have a 315Lg that I'm interested in using via python. I tried the example scripts but they don't work. The viewer and c/c++ examples work fine, so I'm guessing it's not supported via python. I'm actually testing via USB-C currently, not GMSL. Is there a way to use the 315Lg via USB-C in python? Or is this planned at some point?

zhonghong322, 2025-02-18:
We don't have a 315Lg device. Are you referring to the 335Lg? You can use the [v2-main branch of pyorbbecsdk](https://github.com/orbbec/pyorbbecsdk/tree/v2-main), which supports the 335Lg.

The 335lg requires hardware and platform driver support. You can refer to the following documentation, or contact our FAE for assistance.

https://www.orbbec.com/docs/gemini-335lg-platform-support-document/
https://www.orbbec.com/docs/gemini-335lg-recommended-fakra-connector-cable-specifications/
https://orbbec-debian-repos-aws.s3.amazonaws.com/product/GMSL%20EVK%20User%20Manual%28precompiled%20driver%20version%29.pdf

peter-mitrano-bg, 2025-02-21:
Yes sorry I meant the 335Lg. We've moved on to using other cameras, but thank you for the information.

closed 2025-02-21 by peter-mitrano-bg