orbbec/pyorbbecsdk #115: AttributeError: 'pyorbbecsdk.pyorbbecsdk.VideoStreamProfile' object has no attribute 'get_extrinsic_to'

shyzhxpyl, 2025-03-06:
1. 相机：335L
2. 固件：1.2.81
3. 代码版本：v2.0.9
4. 问题描述，执行下面代码，报错AttributeError: 'pyorbbecsdk.pyorbbecsdk.VideoStreamProfile' object has no attribute 'get_extrinsic_to'
```
 try:
      left_profile_list = self.pipeline.get_stream_profile_list(OBSensorType.LEFT_IR_SENSOR)
      right_profile_list = self.pipeline.get_stream_profile_list(OBSensorType.RIGHT_IR_SENSOR)
      depth_profile_list = self.pipeline.get_stream_profile_list(OBSensorType.DEPTH_SENSOR)
      color_profile_list = self.pipeline.get_stream_profile_list(OBSensorType.COLOR_SENSOR)
      try:
          left_ir_profile = left_profile_list.get_video_stream_profile(1280, 800, OBFormat.Y8, 30)
          right_ir_profile = right_profile_list.get_video_stream_profile(1280, 800, OBFormat.Y8, 30)
      except OBError as e:
          print(e)
          left_ir_profile = left_profile_list.get_video_stream_profile(1280, 800, OBFormat.Y8)
          right_ir_profile = right_profile_list.get_video_stream_profile(1280, 800, OBFormat.Y8)
  except Exception as e:
      print(e)
      return
  left_intrinsic = left_ir_profile.get_intrinsic()
  right_intrinsic = right_ir_profile.get_intrinsic()
  left_dis = left_ir_profile.get_distortion()
  right_dis = right_ir_profile.get_distortion()
  print(left_ir_profile.__dir__())
  left_extrinsic = left_ir_profile.get_extrinsic_to(right_ir_profile)
  right_extrinsic = right_ir_profile.get_extrinsic_to(left_ir_profile)
```
同时print(left_ir_profile.__dir__())输出：['__init__', '__doc__', '__module__', 'get_width', 'get_height', 'get_fps', 'get_intrinsic', 'get_distortion', '__repr__', 'get_format', 'get_type', 'is_video_stream_profile', 'is_acce
l_stream_profile', 'is_gyro_stream_profile', 'as_video_stream_profile', 'as_accel_stream_profile', 'as_gyro_stream_profile', '__new__', '__hash__', '__str__', '__getattribute__', '__se
tattr__', '__delattr__', '__lt__', '__le__', '__eq__', '__ne__', '__gt__', '__ge__', '__reduce_ex__', '__reduce__', '__subclasshook__', '__init_subclass__', '__format__', '__sizeof__', '__dir__', '__class__']，确实没有get_extrinsic_to方法
* 查看源代码，看起来是有get_extrinsic_to这个方法的，其他方法都能对的上，只有这个方法没有导出，不知道是什么原因
```
void define_stream_profile(const py::object &m) {
  py::class_<ob::StreamProfile, std::shared_ptr<ob::StreamProfile>>(
      m, "StreamProfile")
      .def("get_format",
           [](const std::shared_ptr<ob::StreamProfile> &self) {
             return self->format();
           })
      .def("get_type",
           [](const std::shared_ptr<ob::StreamProfile> &self) {
             return self->type();
           })
      .def("is_video_stream_profile",
           [](const std::shared_ptr<ob::StreamProfile> &self) {
             return self->is<ob::VideoStreamProfile>();
           })
      .def("is_accel_stream_profile",
           [](const std::shared_ptr<ob::StreamProfile> &self) {
             return self->is<ob::AccelStreamProfile>();
           })
      .def("is_gyro_stream_profile",
           [](const std::shared_ptr<ob::StreamProfile> &self) {
             return self->is<ob::GyroStreamProfile>();
           })
      .def("as_video_stream_profile",
           [](std::shared_ptr<ob::StreamProfile> &self) {
             OB_TRY_CATCH({
               if (!self->is<ob::VideoStreamProfile>()) {
                 throw std::invalid_argument("Not a video stream profile");
               }
               return self->as<ob::VideoStreamProfile>();
             });
           })
      .def("as_accel_stream_profile",
           [](std::shared_ptr<ob::StreamProfile> &self) {
             OB_TRY_CATCH({
               if (!self->is<ob::AccelStreamProfile>()) {
                 throw std::invalid_argument("Not an accel stream profile");
               }
                 return self->as<ob::AccelStreamProfile>();
             });
           })
      .def("as_gyro_stream_profile",
           [](std::shared_ptr<ob::StreamProfile> &self) {
             OB_TRY_CATCH({
               if (!self->is<ob::GyroStreamProfile>()) {
                 throw std::invalid_argument("Not a gyro stream profile");
               }
               return self->as<ob::GyroStreamProfile>();
             });
           })
      .def("get_extrinsic_to",
           [](const std::shared_ptr<ob::StreamProfile> &self,
              const std::shared_ptr<ob::StreamProfile> &target) {
             OB_TRY_CATCH({
               return self->getExtrinsicTo(target);
             });
           });
}

void define_video_stream_profile(const py::object &m) {
  py::class_<ob::VideoStreamProfile, ob::StreamProfile,
             std::shared_ptr<ob::VideoStreamProfile>>(m, "VideoStreamProfile")
      .def("get_width",
           [](const std::shared_ptr<ob::VideoStreamProfile> &self) {
             return self->width();
           })
      .def("get_height",
           [](const std::shared_ptr<ob::VideoStreamProfile> &self) {
             return self->height();
           })
      .def("get_fps",
           [](const std::shared_ptr<ob::VideoStreamProfile> &self) {
             return self->fps();
           })
      .def("get_intrinsic",
           [](const std::shared_ptr<ob::VideoStreamProfile> &self) {
             return self->getIntrinsic();
           })
      .def("get_distortion",
           [](const std::shared_ptr<ob::VideoStreamProfile> &self) {
             return self->getDistortion();
           })
      .def("__repr__", [](const std::shared_ptr<ob::VideoStreamProfile> &self) {
        return "<VideoStreamProfile: " + std::to_string(self->width()) + "x" +
               std::to_string(self->height()) + "@" +
               std::to_string(self->fps()) + ">";
      });
}
```

YeWenxuan64, 2025-03-06:
是自编译的还是用现成的wheel？

shyzhxpyl, 2025-03-11:
> 是自编译的还是用现成的wheel？

自己编译的

YeWenxuan64, 2025-03-26:
use the 'libOrbbecSDK.so.2' in whl of arrch64 to replace the damage file in pyorbbecsdk/sdk/lib/arm64/ , then build again

[image]