orbbec/pyorbbecsdk #116: 2.0.9 doesn't build properly

dbestPacksol, 2025-03-06:
It appears that latest updates in the repo have broken the build scripts. Most noticeably is that cmake is unable to find the orbbecsdk files in the latest update.

YeWenxuan64, 2025-03-06:
actually they missed to upload it.................

YeWenxuan64, 2025-03-26:
use the 'libOrbbecSDK.so.2' in whl of arrch64 to replace the damage file in pyorbbecsdk/sdk/lib/arm64/ , then build again

[image]

zhonghong322, 2025-03-27:
sorry ,I will fix it immediately.

YeWenxuan64, 2025-03-27:
> sorry ,I will fix it immediately.

if you have fixed it，you can close my pull request.

zhonghong322, 2025-03-28:
@YeWenxuan64 @dbestPacksol ，  I  Fixed, this commit:  065f3bfa01b4acbd931ad72e9425f580e27e698c