orbbec/pyorbbecsdk #125: DeviceComponentPtr is nullptr

zxyfl, 2025-04-05:
I use pyinstaller  to make the depth_view.py as a exe .then cmd show the error DeviceComponentPtr is nullptr.

zxyfl, 2025-05-25:
> Hello,I have same Issue,have you solved this problem?

没有解决，问了官方技术，方案是要用1.3.4的版本自己编译

发自我的iPhone

dumhdurum419, 2025-05-25:
okay, thanks for your reply.

huqng, 2026-05-24:
I met a similar problem when packing my code with pyinstaller, but at initialization, with exception `deviceComponentPtr is nullptr`.

Here's a solution given by [deepseek](https://platform.deepseek.com/) and it worked for me. That is, you need to manually copy `$pyorbbecsdkDir/extensions/` to your target directory `$distpath/$appname/_internal/pyorbbecsdk/`, as pyinstaller couldn't recognize the dependencies even with `--collect-all`.