orbbec/pyorbbecsdk #127: Gemini 335L 有没有办法在使用align_filter_viewer.py或者类似方法的时候让深度图与色彩图在时间上能够重合？

Kk-Ling, 2025-04-08:
[image]

深度图总是比彩色图慢，能否通过已有办法解决？否则我只能通过延迟显示色彩图来实现，这样就达不到实时的效果

zhonghong322, 2025-05-06:
Please use the following two samples for testing.
https://github.com/orbbec/pyorbbecsdk/blob/v2-main/examples/hw_d2c_align.py
https://github.com/orbbec/pyorbbecsdk/blob/v2-main/examples/sync_align.py

Kk-Ling, 2025-05-07:
> Please use the following two samples for testing. https://github.com/orbbec/pyorbbecsdk/blob/v2-main/examples/hw_d2c_align.py https://github.com/orbbec/pyorbbecsdk/blob/v2-main/examples/sync_align.py

If the examples run directly, the color will still be real-time, and the depth will still have delay. Ive tried all the samples given and still cannot solve my problem.

zhonghong322, 2025-05-07:
Check the timestamps of depth and color frames in the frameset — if the difference is very small, then the delay is likely due to the device's output. On my side, depth and color are synchronized correctly during testing.

What is the firmware version of your device and which version of the Python SDK are you using?

Kk-Ling, 2025-05-16:
> Check the timestamps of depth and color frames in the frameset — if the difference is very small, then the delay is likely due to the device's output. On my side, depth and color are synchronized [...]

The delay of depth is exactly 5 fames at 10 fps and 2 frames in 5 fps. 
firmware version: 1.2.81
Python SDK version: 2.0.9

zhonghong322, 2025-06-26:
> 我们也在Win10平台遇到了深度图延迟的问题。 偶然发现，运行 OrbbecViewer 后，延迟现象就消失了。似乎每次在新Windows环境配置后都会出现延迟，但只要启动OrbbecViewer，问题就解决了。您可以尝试一下。
> 
> OrbbecViewer下载链接：https://github.com/orbbec/OrbbecSDK/releases
> 
> 期待官方的答复。

@degts @Kk-Ling  Thank you for your feedback on the issue. This is because metadata needs to be registered on Windows. You can follow [the script](https://github.com/orbbec/OrbbecSDK_v2/blob/main/scripts/env_setup/obsensor_metadata_win10.md ) in the README to register it. Later, I will copy this script to the Python wrapper and update the Python documentation accordingly.

When you run OrbbecViewer v1.x.x, the tool will automatically register the metadata by default upon opening. Therefore, it works as expected. If the metadata is not registered, there will be timestamp anomalies, which is the reason for the issue you reported.

closed 2025-07-03 by zhonghong322