orbbec/pyorbbecsdk #128: ImportError: DLL load failed while importing pyorbbecsdk: The parameter is incorrect.

KRS12345678, 2025-04-09:
Help, I have no idea what is wrong

ojas200, 2025-04-27:
Hey. I'd been facing the same issue.

Can you confirm if you've run the pyorbbecsdk install and setup files within the same directory, prior to running this command? Adding the [link](https://github.com/orbbec/pyorbbecsdk/blob/main/docs/README_EN.md) for your reference.

I think you're trying to set up on Windows. Have you tried building and running on Ubuntu? It would be easier to pinpoint the problem.

Also, can you provide the terminal output of the following command:
`python -c "import struct; print(struct.calcsize('P') * 8)"`

KRS12345678, 2025-05-01:
Hey @ojas200 thanks for the reply! I am currently running on Linux now and am running into a different problem. When I try to run pointcloud_filter_o3d.py I get an error saying "No module named 'open3d'. However when I run pip3 list open3d is clearly listed as an installed module. Wondering if you would have any insight on this problem.

KRS12345678, 2025-05-01:
@ojas200 some further info on the above problem. Now I am running into "ImportError: /home/research/pyorbbecsdk/venv/lib/python3.11/site-packages/open3d/linux/open3d.so: cannot open shared object file: No such fil or directory" However the file is there so im not sure whats up