orbbec/pyorbbecsdk #131: Unsupported format for C2D conversion

jonas87, 2025-04-13:
**Describe the bug**  
On a Gemini 330 most C2D conversions don't seem to work. D2C works ok.

**Reproduction Steps**  
Steps to reproduce the behavior:

for example when setting up the streams as:
color_profile = color_profile_list.get_video_stream_profile(640, 480, ob.OBFormat.MJPG, 15)
depth_profile = depth_profile_list.get_video_stream_profile(640, 480, ob.OBFormat.Y16, 15)

and the alignment filter:
align_filter = ob.AlignFilter(align_to_stream=ob.OBStreamType.DEPTH_STREAM)

this results in error when calling the filter:
[04/13 23:51:05.315733][error][22336][AlignImpl.cpp:702] Unsupported format for C2D conversion yet!

**Expected behavior**  
Same config but aligning to the COLOR_STREAM (D2C) works correctly.
Setting the color video stream to OBFormat.RGB instead OBFormat.MJPG makes it work for C2D; but I need MJPG because I actually want to use the higher resolutions.

So I think C2D doesn't work when the color stream is in MJPG

But It actually works in Orbbec Viewer:
In the Point Cloud menu with setting:
Align Mode: C2D(SW)
Color Stream Profile: 640x480 15 fps MJPG
Depth Stream Profile: 640x480 15 fps Y16
no problems!

**Log Output**  
[04/13 23:51:05.315733][error][22336][AlignImpl.cpp:702] Unsupported format for C2D conversion yet!

**System Environment:**

- OS: Windows 11, 
- Architecture: x86_64, 

**Tool/SDK Version:**

- OrbbecViewer：: Version 1.10.16

**Device Information:**

- Model: Gemini 335
- Firmware Version: 1.4.60

zhonghong322, 2025-05-06:
color_profile = color_profile_list.get_video_stream_profile(640, 480, OBFormat.RGB, 15)

we recommend using the [v2-main branch](https://github.com/orbbec/pyorbbecsdk/releases/tag/v2.0.10), which is a Python wrapper based on the open-source SDK.

zhonghong322, 2025-05-07:
> > color_profile = color_profile_list.get_video_stream_profile(640, 480, OBFormat.RGB, 15)
> > We recommend using the [v2-main branch](https://github.com/orbbec/pyorbbecsdk/releases/tag/v2.0.10), which is a Python wrapper based on the open-source SDK.
> 
> Hi zhonghong, thank you for your reply. I was already using the V2 branch. I now installed the latest version pyorbbecsdk-2.0.10 and the problem is still present when using C2D when aligning MJPG stream to Y16 depth stream.
> 
> [05/06 20:55:17.575013][error][23812][AlignImpl.cpp:1529] Unsupported format for C2D conversion yet!

C2D does not support the MJPEG format， so you must set OBFormat.RGB
color_profile = color_profile_list.get_video_stream_profile(640, 480, OBFormat.RGB, 15)

~~~
# ******************************************************************************
#  Copyright (c) 2024 Orbbec 3D Technology, Inc
#
#  Licensed under the Apache License, Version 2.0 (the "License");
#  you may not use this file except in compliance with the License.
#  You may obtain a copy of the License at
#
#      http:# www.apache.org/licenses/LICENSE-2.0
#
#  Unless required by applicable law or agreed to in writing, software
#  distributed under the License is distributed on an "AS IS" BASIS,
#  WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
#  See the License for the specific language governing permissions and
#  limitations under the License.
# ******************************************************************************
import argparse
import sys

import cv2
import numpy as np

from pyorbbecsdk import *
from utils import frame_to_bgr_image

ESC_KEY = 27

MIN_DEPTH = 20  # 20mm
MAX_DEPTH = 10000  # 10000mm
# Temporal filter for smoothing depth data over time

def main(argv):
    pipeline = Pipeline()
    config = Config()
    parser = argparse.ArgumentParser()
    parser.add_argument("-s", "--enable_sync", help="enable sync", type=bool, default=True)
    args = parser.parse_args(argv)

    enable_sync = args.enable_sync
    try:
        profile_list = pipeline.get_stream_profile_list(OBSensorType.COLOR_SENSOR)
        color_profile = profile_list.get_video_stream_profile(640, 480, OBFormat.RGB, 15)
        config.enable_stream(color_profile)
        profile_list = pipeline.get_stream_profile_list(OBSensorType.DEPTH_SENSOR)
        depth_profile = profile_list.get_video_stream_profile(640, 480, OBFormat.Y16, 15)
        config.enable_stream(depth_profile)
    except Exception as e:
        print(e)
        return

    if enable_sync:
        try:
            pipeline.enable_frame_sync()
        except Exception as e:
            print(e)

    try:
        pipeline.start(config)
    except Exception as e:
        print(e)
        return

    align_filter = AlignFilter(align_to_stream=OBStreamType.DEPTH_STREAM)

    while True:
        try:
            frames = pipeline.wait_for_frames(100)
            if not frames:
                continue
            color_frame = frames.get_color_frame()
            depth_frame = frames.get_depth_frame()
            if not color_frame or not depth_frame:
                continue
            frames = align_filter.process(frames)
            if not frames:
                continue
            frames  = frames.as_frame_set()
            color_frame = frames.get_color_frame()
            depth_frame = frames.get_depth_frame()
            if not color_frame or not depth_frame:
                continue

            color_image = frame_to_bgr_image(color_frame)
            if color_image is None:
                print("Failed to convert frame to image")
                continue
            try:
                depth_data = np.frombuffer(depth_frame.get_data(), dtype=np.uint16).reshape(
                    (depth_frame.get_height(), depth_frame.get_width()))
            except ValueError:
                print("Failed to reshape depth data")
                continue
            depth_data = depth_data.astype(np.float32) * depth_frame.get_depth_scale()
            depth_data = np.where((depth_data > MIN_DEPTH) & (depth_data < MAX_DEPTH), depth_data, 0)
            depth_data = depth_data.astype(np.uint16)
            depth_image = cv2.normalize(depth_data, None, 0, 255, cv2.NORM_MINMAX)
            depth_image = cv2.applyColorMap(depth_image.astype(np.uint8), cv2.COLORMAP_JET)
            depth_image = cv2.addWeighted(color_image, 0.5, depth_image, 0.5, 0)

            cv2.imshow("SyncAlignViewer", depth_image)
            if cv2.waitKey(1) in [ord('q'), ESC_KEY]:
                break
        except KeyboardInterrupt:
            break
    cv2.destroyAllWindows()
    pipeline.stop()

if __name__ == "__main__":
    main(sys.argv[1:])

~~~

jonas87, 2025-05-07:
> C2D does not support the MJPEG format, so you must set OBFormat.RGB.

Is this planned for the future? I would like to perform C2D at the highest 16:9 resolution the depth camera supports, 1280x720, but OBFormat.RGB doesn't work at that resolution, it only goes up to 640x480, the higher resolution 1280x720 only works with OBFormat.MJPG.

By the way the Orbbec Viewer does seem support C2D with MJPG, so it's only missing in the python SDK?

zhonghong322, 2025-05-08:
- Is this planned for the future?
  It won't be supported in the future.

- I would like to perform C2D at the highest 16:9 resolution the depth camera supports, 1280x720, but OBFormat.RGB doesn't work at that resolution, it only goes up to 640x480, the higher resolution 1280x720 only works with OBFormat.MJPG.

the highest 16:9 resolution is support in RGB,for example.
~~~
        profile_list = pipeline.get_stream_profile_list(OBSensorType.COLOR_SENSOR)
        color_profile = profile_list.get_video_stream_profile(1280, 720, OBFormat.RGB, 15)
        config.enable_stream(color_profile)
        profile_list = pipeline.get_stream_profile_list(OBSensorType.DEPTH_SENSOR)
        depth_profile = profile_list.get_video_stream_profile(848, 480, OBFormat.Y16, 15)
        config.enable_stream(depth_profile)
~~~

- By the way the Orbbec Viewer does seem support C2D with MJPG, so it's only missing in the python SDK?
  The OrbbecViewer first converts MJPEG to RGB and then feeds it into the C2D filter.

jonas87, 2025-05-08:
> the highest 16:9 resolution is support in RGB,for example.

Hi Zhong, you are right, the mistake I made with get_video_stream_profile() was using OBFormat.RGB which results in:
pyorbbecsdk.OBError: Invalid input, No matched video stream profile found!

I assumed 1280, 720 wasn't supported, but the mistake is that it should be  OBFormat.**BGR** not RGB

Thanks again for your support!

closed 2025-05-08 by jonas87

hongsamvo, 2025-10-01:
Hi @zhonghong322, I am just test your sample code and compare with results from hw_dc_align.py from lastest version and I see the significant different. The result from your sample code is much better. Can you explain this difference?