orbbec/pyorbbecsdk #134: Support for Streaming RGB Data from Gemini 2XL Sensor using Python

AmalNLal, 2025-05-07:
Hi team,

I’m working with the Gemini 2XL sensor and would like to stream RGB data using Python. However, I’m having trouble identifying the correct SDK and whether there is an officially supported Python wrapper.

Sensor Model: Gemini 2XL
Goal: Stream RGB data in real-time using Python

Request:
1. Which SDK should be used to interface with the Gemini 2XL sensor?
2. Is there an official or recommended Python wrapper for this SDK?
3. Are there any example scripts or documentation available for streaming RGB data using Python?

Any guidance or resources would be greatly appreciated. Thank you for your support!

zhonghong322, 2025-05-08:
you can use [main branch](https://github.com/orbbec/pyorbbecsdk/tree/main)

arymandeshwal, 2025-05-19:
I am also facing a similar issue. When i run python3 examples/net_device.py it throws an error saying:
OBError no supported device found at 192.168.1.10:8090 
I am also using Gemini 2xl