orbbec/pyorbbecsdk #139: When will numpy2.0 be compatible？什么时候才能兼容版本为2.0及以上的numpy

YeWenxuan64, 2025-05-11:
我这里有需要用到numpy2.0的工程，可是pyorbbecsdk2还是会在使用numpy2及以上版本时报错。。。

YeWenxuan64, 2025-05-16:
[image]
pybind11都支持numpy2了，你们加点油ヽ(*。>Д<)o゜

closed 2025-08-09 by YeWenxuan64

Rainerino, 2025-09-29:
> 我把requirement里面的pybind11版本限制去掉，把pybind11升级到最新再编译pyobsdk，好像能用numpy2了。（Gemini336L部分功能）

I built the package with the newest pybind11 and numpy, didn't run into any problems when running the example (yet to test  all the features). What's stopping you from migrating to numpy>=2? 

```bash
numpy           2.2.6
opencv-python   4.12.0.88
pybind11        3.0.1
pybind11-global 3.0.1
pygame          2.6.1
pynput          1.8.1
pyorbbecsdk     2.0.13
python-xlib     0.33
```