orbbec/pyorbbecsdk #147: Issue connecting to Gemini 335Le over network by IP

dbestPacksol, 2025-06-06:
[image]

I have the following issue when attempting to connect to a Gemini 335Le over the network when creating via context.create_net_device with it's IP. The IP address is correct as it is the same in OrbbecViewer and can be connected to.

[image]

What am I missing to be able to do this?

gongye2, 2026-10-10:
Hi @dbestPacksol , apologies for the confusion — my previous reply was incorrect.

You're right. The Gemini 335Le currently does not support being opened via `cxt.create_net_device(ip, port)` ; it can only be opened after successful GVCP broadcast discovery. The earlier statement that upgrading to pyorbbecsdk v2.0.13 / OrbbecSDK v2.4.8 would enable this for Gemini 335Le was wrong.

For this specific issue, upgrading to pyorbbecsdk v2.0.13 will not resolve it.

We'll investigate a solution (either opening from a known IP directly, or adding unicast discovery support) and will update this issue once we have a conclusion.

Thanks for the correction, and sorry again for the misleading information.

monforte-dt, 2026-10-16:
I am having the same issue while using orbbec sdk 2.1.2

Weirdly it works on arm64 machines, but not on my x86 ones.

I tried with multiple python versions.