orbbec/pyorbbecsdk #161: 如何设置pipeline获取imu帧的配置

YeWenxuan64, 2025-08-09:
config = Config()
pipeline = Pipeline()

config.enable_accel_stream()
config.enable_gyro_stream()

config.set_frame_aggregate_output_mode(OBFrameAggregateOutputMode.FULL_FRAME_REQUIRE)
pipeline.start(config)

如何设置imu采样速率之类的，手动设置到50hz，200hz之类的

gongye2, 2025-08-28:
You could configure it like this. The full-scale range depends on the device type, while the sample rate must be set for both the accelerometer and the gyroscope and they must be the same; otherwise, the configuration may not take effect as expected:
config.enable_accel_stream(OBAccelFullScaleRange.ACCEL_FS_4g, OBGyroSampleRate.SAMPLE_RATE_50_HZ)
config.enable_gyro_stream(OBGyroFullScaleRange.FS_1000dps, OBGyroSampleRate.SAMPLE_RATE_50_HZ)

YeWenxuan64, 2025-09-02:
> You could configure it like this. The full-scale range depends on the device type, while the sample rate must be set for both the accelerometer and the gyroscope and they must be the same; [...]

great😆